thesofproject/sof #6: Trace mailbox shows garbage

lgirdwood, 2018-06-21:
The trace mailbox is outputting garbage data after patch 80f3d47215c67afb0036b09b849a12799e584254 was applied. 

0x0 [6076024018.780075]	delta [********]	value 0x03a3b651fd7d6ccc
0x10 [26736773379.643696]	delta [********]	value 0x0f2eac82f5e4a85a
0x20 [304907117229.521912]	delta [********]	value 0xe5847e76513e5b06
0x30 [922837080142.584229]	delta [********]	value 0xfae37d57071fc582
0x40 [951347671207.020264]	delta [********]	value 0xff17f99f019e7557
0x50 [451660774.405905]	delta [********]	value 0x00bfe0f6ff85d41d
0x60 [4067790368.758392]	delta [********]	value 0x0132bf27fed5e827
0x70 [4336307921.055160]	delta [********]	value 0x0102b78afee84ac4

Are we missing some PM or cache writeback ?

lgirdwood, 2018-06-21:
Can be reproduced by running "rmbox" on the host. Expectation is that SRAM window may contain 1 or 2 error message but that's all. The rest should be all 0s.

lgirdwood, 2018-06-22:
Confirmed on my UP^2.

closed 2018-06-22 by lgirdwood

keqiaozhang, 2018-07-20:
This issue is fixed by https://github.com/thesofproject/sof/pull/11