thesofproject/sof #7: Firmware crash when attempting playback on UP^2

lgirdwood, 2018-06-22:
Audio playback seems to be broken on my up squared due to the this
commit: 7c281d3d894ce8e9b9ea5a6a314c5c9d8e631d3e. 

I get "unable to install hw_params" error when I start aplay. 

Please find attached the dmesg log below.

[   30.124183] sof-audio sof-audio: pfn i 15 idx 37 pfn 17dc7f
[   30.124203] sof-audio sof-audio: period_bytes:0x3f00
[   30.124210] sof-audio sof-audio: stream_tag 1
[   30.432301] sof-audio sof-audio: error: ipc timed out for 0x60010000
size 0x60
[   30.432755] sof-audio sof-audio: status: fw entered - code 00000005
[   30.433101] sof-audio sof-audio: error: runtime exception
[   30.433386] sof-audio sof-audio: error: trace point 00004000
[   30.433684] sof-audio sof-audio: error: DSP Firmware Oops
[   30.433970] sof-audio sof-audio: error: Exception Cause:
LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore
access
[   30.434600] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR
0x00000040 PS       0x00070025 SAR     0x00000010
[   30.435123] sof-audio sof-audio: EPC1     0xbe01afc0
EPC2     0xbe01afbd EPC3     0xbe010b5e EPC4    0x00000000
[   30.435645] sof-audio sof-audio: EPC5     0xbe01afbd
EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   30.436168] sof-audio sof-audio: EPS2     0x00060120
EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00060120
[   30.436691] sof-audio sof-audio: EPS6     0x00000000
EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[   30.437214] sof-audio sof-audio: stack dump from 0xbe07ff40
[   30.437509] sof-audio sof-audio: 0xbe07ff40: 0xbe01afc0 0x00000000
0x00000010 0x00000000
[   30.437951] sof-audio sof-audio: 0xbe07ff44: 0xbe044024 0x00000000
0x00000008 0x00000000
[   30.438372] sof-audio sof-audio: 0xbe07ff48: 0x0000000d 0x00000000
0xbe020ae9 0xbe020aef
[   30.438794] sof-audio sof-audio: 0xbe07ff4c: 0xbe010b5b 0x00000000
0xbe046aa4 0x00000000
[   30.439216] sof-audio sof-audio: 0xbe07ff50: 0x00000001 0x00000000
0x00000002 0x00000000
[   30.439638] sof-audio sof-audio: 0xbe07ff54: 0x7e0104c6 0x00000000
0x00004000 0x00000000
[   30.440059] sof-audio sof-audio: 0xbe07ff58: 0x00000008 0x00000000
0xbe041d64 0xbe051800
[   30.440482] sof-audio sof-audio: 0xbe07ff5c: 0x00000000 0x00000000
0x01000000 0x00000000
[   30.440903] sof-audio sof-audio: error: waking up any trace sleepers
[   30.441289] sof-audio sof-audio: error: got wrong posn offset
0xffffffff for PCM -110
[   30.441714] sof-audio sof-audio: ASoC: sof-audio hw params failed:
-110
[   30.442367] sof-audio sof-audio: pcm: free stream 0 dir 0
[   30.752315] sof-audio sof-audio: error: ipc timed out for 0x60030000
size 0xc
[   30.752752] sof-audio sof-audio: status: fw entered - code 00000005
[   30.753098] sof-audio sof-audio: error: runtime exception
[   30.753384] sof-audio sof-audio: error: trace point 00004000
[   30.753683] sof-audio sof-audio: error: DSP Firmware Oops
[   30.753968] sof-audio sof-audio: error: Exception Cause:
LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore
access
[   30.754617] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR
0x00000040 PS       0x00070025 SAR     0x00000010
[   30.755141] sof-audio sof-audio: EPC1     0xbe01afc0
EPC2     0xbe01afbd EPC3     0xbe010b5e EPC4    0x00000000
[   30.755664] sof-audio sof-audio: EPC5     0xbe01afbd
EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   30.756187] sof-audio sof-audio: EPS2     0x00060120
EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00060120
[   30.756711] sof-audio sof-audio: EPS6     0x00000000
EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[   30.757234] sof-audio sof-audio: stack dump from 0xbe07ff40
[   30.757529] sof-audio sof-audio: 0xbe07ff40: 0xbe01afc0 0x00000000
0x00000010 0x00000000
[   30.757951] sof-audio sof-audio: 0xbe07ff44: 0xbe044024 0x00000000
0x00000008 0x00000000
[   30.758373] sof-audio sof-audio: 0xbe07ff48: 0x0000000d 0x00000000
0xbe020ae9 0xbe020aef
[   30.758795] sof-audio sof-audio: 0xbe07ff4c: 0xbe010b5b 0x00000000
0xbe046aa4 0x00000000
[   30.759217] sof-audio sof-audio: 0xbe07ff50: 0x00000001 0x00000000
0x00000002 0x00000000
[   30.759639] sof-audio sof-audio: 0xbe07ff54: 0x7e0104c6 0x00000000
0x00004000 0x00000000
[   30.760061] sof-audio sof-audio: 0xbe07ff58: 0x00000008 0x00000000
0xbe041d64 0xbe051800
[   30.760484] sof-audio sof-audio: 0xbe07ff5c: 0x00000000 0x00000000
0x01000000 0x00000000
[   30.760906] sof-audio sof-audio: error: waking up any trace sleepers
[   30.761550] sof-audio sof-audio: pcm: close stream 0 dir 0

lgirdwood, 2018-06-22:
Bisected to commit 14ca1fa3d5a1d7b40bcbfc36d66b0ad96198a21c

keqiaozhang, 2018-06-22:
This regression also happens on CNL-RVP.

xiulipan, 2018-06-22:
The rmbox log is here:
0x0 [53.891054] delta [53.891054]       dai xru
0x10 [53.891885]        delta [0.000831]        dai xru
0x20 [53.891895]        delta [0.000011]        pipe pxr
0x30 [53.889751]        delta [********]        value 0x17707574
0x40 [53.890408]        delta [0.000657]        value 0x17707574
0x50 [53.891054]        delta [0.000646]        dai xru
0x60 [53.891056]        delta [0.000002]        comp Xun
0x70 [53.891058]        delta [0.000002]        value 0x0000000000040000
0x80 [53.891059]        delta [0.000001]        value 0x00000000000000c0
0x90 [53.891066]        delta [0.000007]        pipe tsk
0xa0 [53.891885]        delta [0.000819]        dai xru
0xb0 [53.891886]        delta [0.000002]        comp Xun
0xc0 [53.891887]        delta [0.000001]        value 0x0000000000040000
0xd0 [53.891889]        delta [0.000001]        value 0x00000000000000c0
0xe0 [53.891895]        delta [0.000006]        pipe pxr
0xf0 [53.891897]        delta [0.000001]        pipe cmd
0x100 [53.891898]       delta [0.000002]        host trg
0x110 [53.891901]       delta [0.000002]        host DDi
0x120 [53.891903]       delta [0.000002]        volume trg
0x130 [53.891905]       delta [0.000002]        dai trg
0x140 [53.892884]       delta [0.000979]        ssp tri
0x150 [53.892887]       delta [0.000003]        ssp Ss1
0x160 [53.892888]       delta [0.000002]        ssp Ss2
0x170 [53.892899]       delta [0.000010]        dma LSo
0x180 [53.892912]       delta [0.000013]        pipe pre
0x190 [53.892914]       delta [0.000002]        host pre
0x1a0 [53.892916]       delta [0.000002]        volume pre
0x1b0 [53.892918]       delta [0.000003]        dai pre
0x1c0 [53.892938]       delta [0.000020]        pipe cmd
0x1d0 [53.892940]       delta [0.000002]        host trg
0x1e0 [53.892942]       delta [0.000002]        host DEn
0x1f0 [53.892945]       delta [0.000003]        volume trg
0x200 [53.892947]       delta [0.000002]        dai trg
0x210 [53.892950]       delta [0.000004]        ssp tri
0x220 [53.892952]       delta [0.000002]        ssp sta
0x230 [53.892956]       delta [0.000003]        pipe tsk

xiulipan, 2018-06-22:
After adding some debug trace patch the panic disappear.
It seems we have some potential memory issues.
I remember we have such a regression mouths later and disappear with some new add patches. Need to root cause it now.

xiulipan, 2018-06-22:
in etrace (sudo rmbox) and dma trace (sudo rmbox -t)
value 0x17707574 
shows only with the patch [https://github.com/thesofproject/sof/commit/7c281d3d894ce8e9b9ea5a6a314c5c9d8e631d3e]
But this patch did not add any trace_value in the commit.
After revert this patch this strange trace value also disappear. 
There is definitely some error with this patch.
Hope this info can have some help.

Thanks

xiulipan, 2018-06-23:
confirmed work on my UP2
but still see unknown
trace value 0x17707574

0xaf80 [86.264330]      delta [0.001000]        value 0x17707574
0xaf90 [86.265330]      delta [0.001000]        value 0x17707574
0xafa0 [86.266330]      delta [0.001000]        value 0x17707574
0xafb0 [86.267330]      delta [0.001000]        value 0x17707574

tlauda, 2018-06-23:
0x17 means it is pm_runtime trace message. You just don't have it defined in rmbox.

ranj063, 2018-06-24:
It doesnt work quite as expected
Audio plays only for about 10 seconds and then I get "underrun!!!" with aplay. Here's the dmesg log:
[   47.096639] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 10560 DAI position 1653024
[   47.180437] sof-audio sof-audio: posn mailbox: posn offset is 0xc104c
[   47.180446] sof-audio sof-audio: posn : host 0xe400 dai 0x656280 wall 0x13cd394c
[   47.180461] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 14592 DAI position 1661088
[   47.180486] sof-audio sof-audio: ipc rx: 0x600a0000 done
[   47.180969] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 14592 DAI position 1661088
[   47.215346] sof-audio sof-audio: pstream 0 status 0x24
[   47.264436] sof-audio sof-audio: posn mailbox: posn offset is 0xc104c
[   47.264446] sof-audio sof-audio: posn : host 0x2400 dai 0x65e080 wall 0x13e5d54d
[   47.264461] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 2304 DAI position 1669152
[   47.264485] sof-audio sof-audio: ipc rx: 0x600a0000 done
[   47.264591] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 2304 DAI position 1669152
[   47.348435] sof-audio sof-audio: posn mailbox: posn offset is 0xc104c
[   47.348445] sof-audio sof-audio: posn : host 0x6300 dai 0x665e80 wall 0x13fe714c
[   47.348459] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 6336 DAI position 1677216
[   47.348485] sof-audio sof-audio: ipc rx: 0x600a0000 done
[   47.348593] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 6336 DAI position 1677216
[   47.432432] sof-audio sof-audio: posn mailbox: posn offset is 0xc104c
[   47.432442] sof-audio sof-audio: posn : host 0xa200 dai 0x66dc80 wall 0x14170d4b
[   47.432457] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 10368 DAI position 1685280
[   47.432481] sof-audio sof-audio: ipc rx: 0x600a0000 done
[   47.432584] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 10368 DAI position 1685280
[   47.507140] sof-audio sof-audio: error: DSP trace buffer overflow 3824 bytes. Total messages 28284
[   47.517230] sof-audio sof-audio: ipc rx: 0x90020000 done
[   47.555342] sof-audio sof-audio: pstream 0 status 0x24
[   47.895376] sof-audio sof-audio: pstream 0 status 0x24
[   47.895485] sof-audio sof-audio: posn mailbox: posn offset is 0xc104c
[   47.895492] sof-audio sof-audio: posn : host 0xde00 dai 0x695280 wall 0x1492194c
[   47.895507] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 14208 DAI position 1725600
[   47.895523] sof-audio sof-audio: pcm: trigger stream 0 dir 0 cmd 0
[   47.895557] sof-audio sof-audio: ipc: send 0x60050000
[   47.896336] sof-audio sof-audio: ipc: 0x60050000 succeeded
[   47.896383] sof-audio sof-audio: ipc rx: 0x600a0000 done
[   47.897580] sof-audio sof-audio: pcm: trigger stream 0 dir 0 cmd 1
[   47.897619] sof-audio sof-audio: ipc: send 0x60040000
[   47.897758] sof-audio sof-audio: ipc: 0x60040000 succeeded
[   47.900639] sof-audio sof-audio: cstream 0 status 0x4
[   47.900664] sof-audio sof-audio: error : DSP panic!
[   47.906153] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[   47.906166] sof-audio sof-audio: status: fw entered - code 00000005
[   47.913252] sof-audio sof-audio: error: runtime exception
[   47.919292] sof-audio sof-audio: error: trace point 00004000
[   47.925631] sof-audio sof-audio: error: DSP Firmware Oops
[   47.931671] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   47.945284] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x00000040 PS       0x00070025 SAR     0x00000010
[   47.956568] sof-audio sof-audio: EPC1     0xbe01b018 EPC2     0xbe01b015 EPC3     0xbe010b5e EPC4    0x00000000
[   47.967850] sof-audio sof-audio: EPC5     0xbe01b015 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   47.979121] sof-audio sof-audio: EPS2     0x00060120 EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00060120
[   47.990402] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[   48.001681] sof-audio sof-audio: stack dump from 0xbe07ff40
[   48.007914] sof-audio sof-audio: 0xbe07ff40: 0xbe01b018 0x00000000 0x00000010 0x00000000
[   48.016961] sof-audio sof-audio: 0xbe07ff44: 0xbe044024 0x00000000 0x00000008 0x00000000
[   48.026001] sof-audio sof-audio: 0xbe07ff48: 0x0000000d 0x00000000 0xbe020b49 0xbe020b4f
[   48.035056] sof-audio sof-audio: 0xbe07ff4c: 0xbe010b5b 0x00000000 0xbe046aa4 0x00000000
[   48.044096] sof-audio sof-audio: 0xbe07ff50: 0x00000001 0x00000000 0x00000002 0x00000000
[   48.053140] sof-audio sof-audio: 0xbe07ff54: 0x7e0104c6 0x00000000 0x00004000 0x00000000
[   48.062190] sof-audio sof-audio: 0xbe07ff58: 0x00000008 0x00000000 0x00030082 0xbe051800
[   48.071246] sof-audio sof-audio: 0xbe07ff5c: 0x00000000 0x00000000 0x01000000 0x00000000
[   48.080296] sof-audio sof-audio: error: waking up any trace sleepers
[   48.508274] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 14208 DAI position 1725600
[   48.508728] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 14208 DAI position 1725600
[   48.508760] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 14208 DAI position 1725600
[   48.508783] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 14208 DAI position 1725600
[   49.117187] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 14208 DAI position 1725600
[   49.625667] sof-audio sof-audio: pcm: trigger stream 0 dir 0 cmd 0
[... cut ...]
[   50.604128] sof-audio sof-audio: 0xbe07ff50: 0x00000001 0x00000000 0x00000002 0x00000000
[   50.613168] sof-audio sof-audio: 0xbe07ff54: 0x7e0104c6 0x00000000 0x00004000 0x00000000
[   50.622209] sof-audio sof-audio: 0xbe07ff58: 0x00000008 0x00000000 0x00030082 0xbe051800
[   50.631250] sof-audio sof-audio: 0xbe07ff5c: 0x00000000 0x00000000 0x01000000 0x00000000
[   50.640292] sof-audio sof-audio: error: waking up any trace sleepers
[   50.647617] sof-audio sof-audio: pcm: close stream 0 dir 0

tlauda, 2018-06-25:
What is the format you are playing?
Are you sure the issue is still related to the commit https://github.com/thesofproject/sof/commit/14ca1fa3d5a1d7b40bcbfc36d66b0ad96198a21c?

ranj063, 2018-06-26:
I am playing 16-bit format with a colume pipeline and output format is 24-bit
let me confirm if the issue is due to that commit still

tlauda, 2018-06-26:
@ranj063 Any updates? Still visible on top of the master? From what I'm seeing in the log, it looks like maybe issue #17.

ranj063, 2018-06-26:
@tlauda sorry I didnt time to check this yesterday. I will update today.

ranj063, 2018-06-27:
@tlauda , I've just tested my UP2 and the problem no longer exists and the playback is clean without distortion as well. 

But, I cant play more than a few seconds of audio likely due to #17.

xiulipan, 2018-06-27:
@ranj063 Can you check the 0xbe01b018 in you sof-apl.lst
Where the panic happens?

keqiaozhang, 2018-06-27:
I have tested this issue on my UP2 board， this issue is no longer exists with nocodec tplg and codec tplg. 
For nocodec topology, the ssp waveform is normal with different input frequency of sine.
For codec topology, the sound is clean without errors, but #17  trace buffer overflow issue is still exists.
Test env:
sof-dev: cb05347d9f85e18b414098d512c8287c297933d2
sof-master:599e1800bab31883cf1f9e845279d079d4950c58
soft-master:
For nocodec tplg: reset to 308cf3d and use test-ssp5-I2S-volume-s16le-s16le-48k-24576k-nocodec.tplg
For codec tplg: f756455e and use test-ssp5-mclk-0-I2S-volume-s16le-s16le-48k-24576k-codec.tplg

closed 2018-06-27 by keqiaozhang

xiulipan, 2018-07-24:
fixed by #23