thesofproject/sof #22: pause capture when playback is runing will fail

xiulipan, 2018-06-27:
When runing aplay and arecord same time and pause the capture will cause panic
[65799.112194] sof-audio sof-audio: PCM: stream 0 dir 0 DMA position 11136 DAI position 92688
[65799.151303] sof-audio sof-audio: pcm: trigger stream 0 dir 1 cmd 4
[65799.151347] sof-audio sof-audio: ipc: send 0x60070000
[65799.151521] sof-audio sof-audio: ipc: 0x60070000 succeeded
[65799.151597] sof-audio sof-audio: PCM: stream 0 dir 1 DMA position 7872 DAI position 24192
[65799.154303] sof-audio sof-audio: error : DSP panic!
[65799.154323] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[65799.154395] sof-audio sof-audio: status: fw entered - code 00000005
[65799.154447] sof-audio sof-audio: error: runtime exception
[65799.154456] sof-audio sof-audio: error: trace point 00004000
[65799.154464] sof-audio sof-audio: error: DSP Firmware Oops
[65799.154493] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[65799.154496] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x00000040 PS       0x00070025 SAR     0x00000020
[65799.154499] sof-audio sof-audio: EPC1     0xbe01b1c4 EPC2     0xbe01b1c1 EPC3     0xbe01b1c1 EPC4    0x00000000
[65799.154502] sof-audio sof-audio: EPC5     0xbe01b1c1 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[65799.154505] sof-audio sof-audio: EPS2     0x00060520 EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00060120
[65799.154508] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[65799.154510] sof-audio sof-audio: stack dump from 0xbe07ff40
[65799.154513] sof-audio sof-audio: 0xbe07ff40: 0xbe01b1c4 0x00000000 0x00000020 0x00000000
[65799.154516] sof-audio sof-audio: 0xbe07ff44: 0xbe044024 0x00000000 0x00000008 0x00000000
[65799.154519] sof-audio sof-audio: 0xbe07ff48: 0x0000000d 0x00000000 0xbe020ce9 0xbe020cef
[65799.154522] sof-audio sof-audio: 0xbe07ff4c: 0xbe010b5b 0x00000000 0xbe046aa4 0x00000000
[65799.154525] sof-audio sof-audio: 0xbe07ff50: 0x00000001 0x00000000 0x00000002 0x00000000
[65799.154528] sof-audio sof-audio: 0xbe07ff54: 0x7e0104c6 0x00000000 0x00004000 0x00000000
[65799.154531] sof-audio sof-audio: 0xbe07ff58: 0x00000008 0x00000000 0x00030082 0xbe054800
[65799.154534] sof-audio sof-audio: 0xbe07ff5c: 0x00000000 0x00000000 0x01000000 0x00000000
[65799.154536] sof-audio sof-audio: error: waking up any trace sleepers

xiulipan, 2018-06-27:
rmbox error trace:

0x0 [65793.655709]      delta [********]        dai xro
0x10 [65793.655723]     delta [0.000014]        pipe pxr
0x20 [65793.656288]     delta [0.000564]        pipe pxr
0x30 [65793.656292]     delta [0.000005]        comp CEs
0x40 [65793.656293]     delta [0.000001]        value 0x0000000000000003
0x50 [65793.656295]     delta [0.000002]        pipe eOp
0x60 [65793.656296]     delta [0.000001]        ipc pc0
0x70 [65793.656297]     delta [0.000001]        value 0x000000000000000a
0x80 [65793.656298]     delta [0.000001]        value 0x0000000000000008
0x90 [65793.656300]     delta [0.000001]        pipe px0
0xa0 [65793.656768]     delta [0.000468]        volume xro
0xb0 [65793.656776]     delta [0.000009]        pipe ePU
0xc0 [65793.656778]     delta [0.000001]        value 0x0000000000000004
0xd0 [65793.656779]     delta [0.000001]        pipe pxr

zhigang-wu, 2018-06-27:
where is the panic location based on the *.lst file?

lgirdwood, 2018-06-27:
xiulipan please label this issue. i.e. HW, regression etc

xiulipan, 2018-06-27:
sorry about that. I tried to do the label work but found it is not same as my own repo.
It seems I did not have the access to do the label work.

zhigang-wu, 2018-07-10:
this bug is fixed by this patch:
https://github.com/thesofproject/sof/pull/35

lgirdwood, 2018-07-10:
@keqiaozhang please close if you can confirm fixed

keqiaozhang, 2018-07-13:
verified, this issue is fixed.

closed 2018-07-13 by keqiaozhang

zhigang-wu, 2018-07-20:
fixed by PR #35