thesofproject/sof #25: Python script to read DSP oops

lgirdwood, 2018-06-27:
Write a python script to parse the DSP firmware oops message and then lookup the corresponding EPC C functions from the map/dissassembly files Dsiplay the EPS states too for each level.

Also look up each function and parameters from stack dump using map/disassembly files.

Jocelyn-Li, 2018-07-05:
Suggest to put into SOF 1.3

pratik0509, 2018-08-16:
Hey, I would like to work on this if it is still open?

lgirdwood, 2018-08-16:
@pratik0509 yep, still open. please have a look at the oops messages in some of the sof issues for examples of text format.. The C format is defined in uapi/ipc.h.  You will also need some xtensa ISA documentation to help you understand the DSP registers. Google for "xtensa ISA".

The goal is to read the oops message and provide a stack dump, alongside runlevel,  C code that each EPC register points too. Anything useful for debugging really.

pratik0509, 2018-08-18:
@lgirdwood Sorry missed the mail before. I'll get to it asap :)

pratik0509, 2018-08-22:
@lgirdwood I looked up for Xtensa ISA exception registers and went through the C format in ipc.h and some of the oops messages. What I understood till now is:
Building a tool which helps in debugging the code. The tool will parse the oops messages and try to link it with the function call or line of code which was responsible for it and generate a Trace Stack(if possible).
Sorry if it sounds dumb, but can you be more precise as to what needs to be done exactly? (new developer here)

lgirdwood, 2018-08-22:
@pratik0509 so we want to take the register dump and stack trace addresses and lookup the symbols for those addresses in the firmware disassembly text file (*.dis). i.e. the inputs to the script would be a kernel dmesg and a firmware disassembly file. The outputs would be the call trace (from the stack dump) and the function that each EPC register pointed too.

lgirdwood, 2018-08-22:
@pratik0509 it's maybe worth while installing Docker and building some firmware images using Docker container. See sof.git/scripts/README.docker. This will give you the disassembly files and firmware binaries. The next step would be to take some dmesg files from github SOF issues tracker and use both dmesg and disassembly file to test your scipt. Please note that you will need to build the firmware that matches the commit ID from the dmesg oops mesage otherwise your symbols and addresses wont match.

tlauda, 2018-11-28:
@lgirdwood @mmaka1 Is it already covered by the new logger and FW core dump parsing feature?

lgirdwood, 2018-11-29:
yep, I guess we can close this now.

closed 2018-11-29 by lgirdwood