thesofproject/sof #46: pcm512x codec load fail on APL at UP^2 board

zhuqingliang, 2018-07-04:
sof-dev kernel: f9ab80f566cfcff9e03eaa5e5bbdc510767cd818
sof: c454399e8890c1d7eb007857f0dbe018d244eeb7
soft: 629d2a6bf1d0cf3c5b069d06fd120ed43ef5040f 
using sof-apl-pcm512x.tplg
dmesg log
[    5.472861] bxt-pcm512x bxt-pcm512x: info: override FE DAI link SSP5-Codec
[    5.472873] bxt-pcm512x bxt-pcm512x: ASoC: CPU DAI sof-audio not registered
[    5.472877] bxt-pcm512x bxt-pcm512x: snd_soc_register_card failed -517
[    5.473473] bxt-pcm512x bxt-pcm512x: info: override FE DAI link SSP5-Codec
[    5.473484] bxt-pcm512x bxt-pcm512x: ASoC: CPU DAI sof-audio not registered
[    5.473487] bxt-pcm512x bxt-pcm512x: snd_soc_register_card failed -517
[    5.494185] bxt-pcm512x bxt-pcm512x: info: override FE DAI link SSP5-Codec
[    5.494198] bxt-pcm512x bxt-pcm512x: ASoC: CPU DAI sof-audio not registered
[    5.494211] bxt-pcm512x bxt-pcm512x: snd_soc_register_card failed -517

xiulipan, 2018-07-05:
This bug should be go with soft or sof.
I had a PR for this https://github.com/thesofproject/linux/pull/12
But since we did not have an solution about how to handle the DMIC part, we may need both the PR and some modification on topology.

lgirdwood, 2018-07-05:
Have review comments been addressed for https://github.com/thesofproject/linux/pull/12 ? If so, please push and ping.

xiulipan, 2018-07-06:
Should be fixed with
https://github.com/thesofproject/soft/pull/19 and https://github.com/thesofproject/linux/pull/12
Change both machine driver and topology to fit the newest dai changes.

zhuqingliang, 2018-07-11:
with the environment ：
sof-dev: b87a468a5bf1863b3d371cb9dfe2c718225feb68
sof: 71413cb0050c16e4926c627bd457644e9045e568
soft:47339d2302f445c3b35d50f9720ee649dd2f739b
test-ssp5-mclk-0-I2S-volume-s16le-s16le-48k-24576k-codec.tplg
tplg load successfully .

lgirdwood, 2018-07-11:
@zhuqingliang please close if you are confirming it's resolved.

closed 2018-07-12 by keqiaozhang