thesofproject/sof #48: Volume topology Waveform distortion on CNL.

zhuqingliang, 2018-07-04:
[image]
sof-dev kernel: f9ab80f566cfcff9e03eaa5e5bbdc510767cd818
sof: c454399e8890c1d7eb007857f0dbe018d244eeb7
soft: 629d2a6bf1d0cf3c5b069d06fd120ed43ef5040f
use test-ssp0-mclk-0-I2S-volume-s16le-s16le-48k-24000k-nocodec.tplg
run alsabat check /tmp/bat.* waveform

tlauda, 2018-07-11:
FW built with gcc or xcc?

keqiaozhang, 2018-07-11:
@tlauda 
We use gcc for CNL.

lgirdwood, 2018-07-11:
Looks like the SSP DMA is reading the period written by volume.

zhigang-wu, 2018-07-11:
@lgirdwood 
If the same buffer is read and written, that would means sometimes the DMA IRQ interval became short. or the pipeline is scheduled for a little late. is that correct?

lgirdwood, 2018-07-11:
@zhigang-wu check the r/w pointers for SSP DMA at trigger start.

zhigang-wu, 2018-07-11:
@lgirdwood 
But from the wave displayed, it seems this distortion happened during the playing process.
I agree check the R/W pointer is only choice to confirm this issue. it is the result.
But I am curious how this situation happened. figure out this reason, the bug will be fixed easily.
I think maybe the DMA or pipeline is abnormal sometimes.
Do you think is there any other possibility?
I think maybe this issue would happen on other platform.

tlauda, 2018-07-11:
It's more likely the cache problem.
Can this issue be easily reproduced (e.g. it's only happening for 24 kHz) or it's rather a random problem?

lgirdwood, 2018-07-12:
@zhigang-wu it may also be cache like @tlauda says if this works fine on BYT.

The GP DMA reads the period in blocks into the DMA FIFO before they are sent out to the SSP FIFO. i.e. I've seen it fetch 24 frames, then 16 then 8 (for a 48 frame period) from the period to the DMA FIFO. There can be a race between these DMA reads and any writers into this period, so if you note that the distortion is valid waveform data in the wrong position (and the wrong data sizes align with DMA FIFO fetch sizes then this can imply R/W race).

zhuqingliang, 2018-07-13:
frequence 997 is ok,  frequency 234  or 1234 is error  on latest commit 

2d92796072856e530c3a1bce4f0a9ce2c74692d4

keqiaozhang, 2018-07-16:
@tlauda This issue only happening on CNL.

tlauda, 2018-07-24:
@keqiaozhang 
How long does it take for glitches to appear? I've tried couple of times with ~2 min playbacks and I didn't see any distortion.

keqiaozhang, 2018-07-25:
@tlauda 
This issue still can be reproduced with latest code base(sof-master: 3a29be). I tested on ssp 0 with nocodec tplg, since no ssp0 nocodec tplg under topology/test/, so I generated it myself by adding "simple_test nocodec volume "NoCodec-0" s16le SSP 0 s16le 25 16 2400000 24000000 I2S 0 SIMPLE_TESTS[@]" in tplg-build.sh. This issue happens from beginning to end.
[image]
Could you please share your test steps to see what's the difference between us? Thanks

tlauda, 2018-07-25:
@keqiaozhang 
I'm running alsabat -Dhw:0,0 -f s16_le -r 48000 -c2 -F234 with the same topology as yours and I don't see any glitches. Volume in alsamixer set to 100% for both playback and capture. Are you connecting SSP pins externally?

keqiaozhang, 2018-07-26:
@tlauda 
This issue is related to the compiler and this issue happens only with gcc,  with xcc, there's no such issue.  So this explains what you found last night.

Jocelyn-Li, 2018-07-27:
@tlauda Can this issue be fixed with different gcc compiler option?  Any suggestions?

lgirdwood, 2018-07-30:
@Jocelyn-Li @keqiaozhang the issue is because GCC does not generate optimised HiFi2/3 SIMD instructions for volume control yet. Lets fix later with gcc update.
@keqiaozhang can you enable verbose trace for both xcc and gcc volume and paste the output of a 5 pipeline schedules for both. This way we will see the difference in performance for volume between gcc and xcc atm.

tlauda, 2018-07-30:
@lgirdwood Last time I checked volume module compiled with GCC took ~5 MCPS to execute, so it's very unlikely it's because of lack of SIMD instructions. Maybe there are some additional scheduling issues. I will take a look.

keqiaozhang, 2018-07-30:
@lgirdwood 
Please refer to the attached log files.
[rmbox-trace-gcc.txt](https://github.com/thesofproject/sof/files/2240315/rmbox-trace-gcc.txt)
[rmbox-trace-xcc.txt](https://github.com/thesofproject/sof/files/2240316/rmbox-trace-xcc.txt)

lgirdwood, 2018-07-30:
@tlauda ok, that's good to know, I am working on hifi 2/3 for gcc but it wont be ready for a while.

keqiaozhang, 2018-08-02:
@tlauda 
I did a sweep test today and this issue cannot be reproduced. let's close this bug.
This issue is fixed by PR#177.

closed 2018-08-02 by keqiaozhang