thesofproject/sof #50: Playback and capture's volume kctls are mixed in one page

keqiaozhang, 2018-07-04:
There're different views of kctl in amixer setting, but we found that all kctls of playback and capture are mixed together. For example, in playback view page, we also can found the kctl of capture and vice versa. This is a common issue on each platform.
[image]

lgirdwood, 2018-07-04:
This is an ALSA specific issue that relies on naming strings to help userspace to determine whether a kcontrols playback or capture.
I think here ALSA uses the second string to determine direction, can you try and change the topology to use "PGA4.0 Capture Master Volume" instead ?

xiulipan, 2018-07-05:
@keqiaozhang 
You can take a reference here about the control name
https://www.alsa-project.org/~tiwai/writing-an-alsa-driver/ch06s03.html

lgirdwood, 2018-07-05:
“SOURCE DIRECTION FUNCTION”.  Ok, so we need to change topology. e.g. "PGA4.0 Master Capture Volume" should become "Master Capture Volume PGA4.0". Can you update topology kcontrol builder macro to fix this.

keqiaozhang, 2018-07-09:
@lgirdwood 
Only adjust the string order in topology doesn't help. The root cause may related to the first string. A control is usually defined from the three parts as “SOURCE DIRECTION FUNCTION”.   Source is a string such as “Master”, “PCM”, “CD” and “Line”, so the first string should not be PGA*. Xiuli will help to check how the control name is parsed in kerenl side.

tlauda, 2018-07-14:
Shouldn't this issue be under linux repo instead of sof?

lgirdwood, 2018-07-14:
Closed as should be kernel issue.

closed 2018-07-14 by lgirdwood

Jocelyn-Li, 2018-07-20:
moved https://github.com/thesofproject/linux/issues/43