thesofproject/sof #58: "DSP panic"  when run aplay on GP using tdf8532 codec

zhuqingliang, 2018-07-05:
4.17-dev:   28fc73b252eaa6438e9c10a854f7067db24f07cd
sof:           709d6301e1ab2a0b2d0e45ad3621610a20232f5b 
soft:          600e2692bd336bb3c1ac9154f3bdb60833ffc96b 
aplay -Dhw:0,0 -f s32_le -r 48000 -c4    4channel-32b.wav  
check dmesg
[   46.547323] sof-audio sof-audio: error : DSP panic!
[   46.547342] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[   46.547351] sof-audio sof-audio: status: fw entered - code 00000005
[   46.547382] sof-audio sof-audio: error: runtime exception
[   46.547389] sof-audio sof-audio: error: trace point 00004000
[   46.547397] sof-audio sof-audio: error: DSP Firmware Oops
[   46.547405] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   46.547414] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x00000040 PS       0x00070025 SAR     0x00000010
[   46.547422] sof-audio sof-audio: EPC1     0xbe01c430 EPC2     0xbe01c42d EPC3     0xbe010b56 EPC4    0x00000000
[   46.547430] sof-audio sof-audio: EPC5     0xbe01c42d EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   46.547438] sof-audio sof-audio: EPS2     0x00060120 EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00060120
[   46.547446] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[   46.547453] sof-audio sof-audio: stack dump from 0xbe07ff40
[   46.547462] sof-audio sof-audio: 0xbe07ff40: 0xbe01c430 0x00000000 0x00000010 0x00000000
[   46.547470] sof-audio sof-audio: 0xbe07ff44: 0xbe044024 0x00000000 0x00000008 0x00000000
[   46.547479] sof-audio sof-audio: 0xbe07ff48: 0x0000000d 0x00000000 0xbe021f19 0xbe021f1f
[   46.547488] sof-audio sof-audio: 0xbe07ff4c: 0xbe010b53 0x00000000 0xbe046aa4 0x00000000
[   46.547497] sof-audio sof-audio: 0xbe07ff50: 0x00000001 0x00000000 0xbe046928 0x00000000
[   46.547505] sof-audio sof-audio: 0xbe07ff54: 0x7e0104c6 0x00000000 0x00004000 0x00000000
[   46.547513] sof-audio sof-audio: 0xbe07ff58: 0x00000008 0x00000000 0xedff0080 0xbe041dd4
[   46.547522] sof-audio sof-audio: 0xbe07ff5c: 0x00000000 0x00000000 0x01000000 0x00000000
[   46.547535] sof-audio sof-audio: error: waking up any trace sleepers

zhigang-wu, 2018-07-06:
this information is not enough.
next time please paste the rmbox's log and the *.lst file.
and how long this issue will happen: immediately or for a little while.
did you do the pause/release operation during aplay?

zhigang-wu, 2018-07-06:
this issue should be same as "ALSA XRUN recovery not working".
when the alsa XRUN, it will send out the stop/start command to fw.
the fw can not recover correctly. the panic happens.

lgirdwood, 2018-07-06:
@zhigang-wu it would also be good to avoid the NULL + 0x40 dereference too. i.e. emit trace and return an error if this condition happens rather than deref NULL + 0x40

zhuqingliang, 2018-07-13:
the DSP panic issue when playback is fixed in latest sof-master

closed 2018-07-13 by zhuqingliang

zhigang-wu, 2018-07-13:
@zhuqingliang 
Did you confirm this issue at this case? the alsa xrun happened in the host.
which platform did you confirm this issue?

xiulipan, 2018-07-24:
random bug.
Maybe not fix but did not tested out.