thesofproject/sof #74: 11 samples of SSP waveform point are error on APL with nocodec tplg

zhuqingliang, 2018-07-09:
environment:
sof-dev:   bbe7c7a835b66dfe34eee7feeea78893cbb71a01
sof:      fa48c42abf83155d1bd42fafd4993d9b421ab5e0
soft:     ff8eb10b1877bb0fac8adf0124c36762a5f912fd 

Reproduce Steps:
using test-ssp5-mclk-0-I2S-volume-s16le-s16le-48k-24576k-nocodec.tplg as sof-apl-nocodec.tplg
run : alsabat -Dhw:0,0 -f dat -c2 -r 48000 -F 997 
audacity open /tmp/bat.wav.*
[image]

zhuqingliang, 2018-07-09:
[dmesg.log](https://github.com/thesofproject/sof/files/2176120/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2176121/rmbox.log)

lgirdwood, 2018-07-09:
How can you capture analog data if this is nocodec mode ? It seems like this should be using a topology for a codec.

tlauda, 2018-07-10:
What does alsabat exactly do in terms of FW pipelines?
Does it create both render and capture pipelines and somehow loopbacks it back from SSP?
I'm trying to understand if this is for sure render problem or maybe a capture problem.

keqiaozhang, 2018-07-10:
alsabat will create playback and capture pipeline, then play a specific sine wav and capture it through the loopback. 
Today I used logic pro to capture the data from SDO port, look like the data is normal, so it's most like a capture issue.
[image]

lgirdwood, 2018-07-10:
@keqiaozhang SSP capture FIFO  is probably full of stale data. It could be cleared during trigger by reading the SSP FIFO directly using DSP reads.

Jocelyn-Li, 2018-07-11:
Keqiao, please make sure it is FW issue or driver issue.

tlauda, 2018-07-11:
@zhuqingliang You have used command run : alsabat -Dhw:0,0 -f dat -c2 -r 48000 -F 997, but screen from Audacity says it's 44.1 kHz. Which one is wrong?

lgirdwood, 2018-07-12:
@keqiaozhang, @tlauda   ok, since this is first time only then we should read the SSP FIFO as part of the SSP driver init to clear out any stale samples. Should be a easy fix to try.

tlauda, 2018-07-17:
Please verify on current top of master. I cannot reproduce this problem anymore.

zhuqingliang, 2018-07-18:
We perform test with  e69ebddb4f61a0e7047efb728f7d7525171b6690 . There are a long period of blank bit at the head of bat.wav* .  The reproduce percentage is about 20% to 30%.
reproduce steps ：
1.using test/test-ssp5-mclk-1-I2S-volume-s32le-s32le-48k-24576k-nocodec.tplg
2.reboot 
3. run ` alsabat -Dhw:0,0 -f s32_le -r 48000 -c2 -F300`  once boot up
4. audacity open /tmp/bat.wav.*
[image]

lgirdwood, 2018-07-18:
@zhuqingliang have you verified this delay/race is not coming from alsabat/Linux userspace ?

tlauda, 2018-07-18:
@zhuqingliang Can you provide dmesg log? I'm using the same topology and the same alsabat command, but I cannot reproduce this issue. Tried to reboot and run alsabat ~20 times.

zhuqingliang, 2018-07-19:
If run alsabat several times , the issues will not happen again .

closed 2018-07-19 by zhuqingliang

keqiaozhang, 2018-07-20:
This issue is fixed  by PR#111.