thesofproject/sof #75: can not make firmware

zhuqingliang, 2018-07-10:
commit : 764f9536cdb196d46f653f542a18d43c0fc2fd1d

Reproduce Steps:
./script/xtensa-build-all.sh

error message:
checking whether the C compiler works... no
configure: error: in `/home/user/workspace/reef/sof':
configure: error: C compiler cannot create executables

xiulipan, 2018-07-10:
bisect to #67 
In configure file the compiler check macro is not replaced with AM_* flags and compiler check in configure will fail.

> ac_compile='$CC  -c $CFLAGS $CPPFLAGS conftest.$ac_ext >&5'

jajanusz, 2018-07-10:
It works for APL/CNL for me. Please try setting SUPPORTED_PLATFORMS=(apl cnl) in scripts/xtensa-build-all.sh and let me know what you get. Thanks

keqiaozhang, 2018-07-10:
@jajanusz 
Still not working with exit 77. T
[config.log](https://github.com/thesofproject/sof/files/2179565/config.log)
he config.log is attached.

jajanusz, 2018-07-10:
@keqiaozhang 
Looks like script didn't set XCC, XTOBJCOPY, XTOBJDUMP variables.
Do you have XTENSA_TOOLS_ROOT variable set to directory that has directory structure as expected by script? With folders like:
```
.
└── install
    ├── builds
    │   ├── RF-2016.4-linux
    │   └── RG-2017.8-linux
    └── tools
        ├── RF-2016.4-linux
        └── RG-2017.8-linux
```

jajanusz, 2018-07-10:
@keqiaozhang 
Also please let me know if it build for you without script with something like:
```
./autogen.sh
./configure --with-arch=xtensa --with-platform=apollolake --with-dsp-core=$XTENSA_CORE --with-root-dir=$CONFIG_PATH/xtensa-elf --host=xtensa-bxt-elf --with-meu=$MEU_PATH --with-key=$PRIVATE_KEY_PATH CC=xt-xcc OBJCOPY=xt-objcopy OBJDUMP=xt-objdump
make
make bin
```
With properly set variables of course.

keqiaozhang, 2018-07-10:
@jajanusz 
I have set the XTENSA_TOOLS_ROOT, but nothing help. 
 $ ./configure --with-arch=xtensa --with-platform=cannonlake --with-root-dir=/home/zhangke/xtensa/XtDevTools//install/builds/RF-2016.4-linux/X6H3CNL_2016_4_linux/xtensa-elf --host=xtensa-cnl-elf CC=xt-xcc OBJCOPY=xt-objcopy OBJDUMP=xt-objdump --with-dsp-core=X6H3CNL_2016_4_linux

jajanusz, 2018-07-10:
Can you send config.log for your last command, where you manually set CC=xt-xcc?

jajanusz, 2018-07-10:
I think it's not a problem directly caused by flags (it should pass even with no flags), if you think otherwise you can test it with changing CFLAGS to whatever you think should be there. Do it before autoconf checks first time for compiler, what is probably there: https://github.com/thesofproject/sof/blob/a48ceddc82b6dc2cfd64a622f5c32fc80293692d/configure.ac#L141

C compiler is not checked with **AC_PROG_CC**, as this comment says:
https://github.com/thesofproject/sof/blob/a48ceddc82b6dc2cfd64a622f5c32fc80293692d/configure.ac#L373
So changing flags before it will not help you.

Let me know if chagning CFLAGS helped you, so I'll make patch to make it work for everyone.
Also please send me logs.
Thanks

keqiaozhang, 2018-07-11:
@jajanusz 
I verified the latest code base this morning, now only APL can pass the compilation with xcc. but CNL, this issue still exists.  Please help to address it.
[config.log](https://github.com/thesofproject/sof/files/2182634/config.log)

keqiaozhang, 2018-07-11:
@jajanusz 
Reconfirm the case on CNL. From the config.log you can see that we miss some xtensa config of CNL for xt-xcc. Will retest after get the right version xt-xcc config for CNL.

jajanusz, 2018-07-11:
It looks like problem with your installation/config. I have the same core as yours.
Your log says what error is:
```
You need to either specify the name of a registered Xtensa core (with
the --xtensa-core option or the XTENSA_CORE environment variable) or
specify a different registry of Xtensa cores (with the --xtensa-system
option or the XTENSA_SYSTEM environment variable).
configure:3729: $? = 2
configure:3718: xt-xcc -v >&5
Error: there is no Xtensa core registered as the default.
```
You can for example just set env variable like:
```
export XTENSA_CORE=X6H3CNL_2016_4_linux
```
Let me know if it helps.
Thanks

keqiaozhang, 2018-07-11:
@jajanusz 
Thanks, we have root caused this bug, it's because we don't have X6H3CNL_2016_4_linux, our developer is working on it.

keqiaozhang, 2018-07-12:
summary:
The compilation issue is caused by 17c929e and both gcc and xcc cannot work.
1. gcc compilation issue: let's track it with #78
2. xcc compilation issue: now we only support  APL and CNL platform. For APL, this issue is fixed by Jajanusz, commit ID:de100b7d2823e0496a0cad055625063204d0d585. For CNL, it's because we miss  build "X6H3CNL_2016_4_linux".  so this issue can close now and track the xcc issue on CNL with #93

closed 2018-07-12 by keqiaozhang

keqiaozhang, 2018-07-20:
This issue is fixed by PR#77.