thesofproject/sof #79: IPC timed out after system is booted

keqiaozhang, 2018-07-11:
This issue is caused by PR#71:trace: fix dma trace with repeated data bug on APL&CNL.
Due to the compilation issue #78 and #75, I only found this issue on APL. Will check on other platforms after the compilation issues are fixed.
[dmesg.log](https://github.com/thesofproject/sof/files/2183091/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2183092/rmbox.log)

RanderWang, 2018-07-11:
I also checked it at commit: fa48c42abf which my patch is based on at the day I made PR, not such issue. I think it is not caused by my patch but I will help to fix it.

xiulipan, 2018-07-11:
should be fixed by #82 
@keqiaozhang  could you test to confirm

keqiaozhang, 2018-07-12:
This issue is fixed.

closed 2018-07-12 by keqiaozhang

keqiaozhang, 2018-07-20:
This issue is fixed  by https://github.com/thesofproject/sof/pull/82