thesofproject/sof #101: Stress test: pause/release playback or capture  when both playback and capture are running will cause IPC timed out error

zhuqingliang, 2018-07-13:
environment:
on UP^2
sof-dev: a3a39f21218c074c71cd28b0c2c782b1626df3e3
sof: 2d92796072856e530c3a1bce4f0a9ce2c74692d4
soft: 47339d2302f445c3b35d50f9720ee649dd2f739b
tplg file name :
topology/test/test-ssp5-mclk-0-I2S-volume-s16le-s16le-48k-24576k-nocodec.tplg

reproduce steps:
run aplay and arecord at same time , pressing space key ,don't release . A few seconds later ' input/output error' . check dmesg log.

log:

[dmesg6.log](https://github.com/thesofproject/sof/files/2193081/dmesg6.log)
[rmbox6.log](https://github.com/thesofproject/sof/files/2193082/rmbox6.log)

zhigang-wu, 2018-07-16:
the initial analysis for this log:
1. you press the space key for capture feature to trigger this issue.
2. the firmware's xrun is happened first, then ipc timeout.
3. the timestamp between two log file are not conformance, rmbox's log has the issue is at about 8502s, the dmesg's log has the issue is at about 8507s.
4. the first part of log in rmbox, I analysis here:
0x0 [8502.764351]	delta [********]	value 0x00000000000a0000
0x10 [8502.764353]	delta [0.000001]	value 0x00000000000000c0
0x20 [8502.764357]	delta [0.000004]	[31mipc eQb[0m
                   seems fw tries to send the xrun posn info to host, but failed for lacking of message header,
                   the previous xrun log is missed here.
0x30 [8502.764365]	delta [0.000009]	pipe pxr
                   pipeline recover process start.
0x40 [8502.764367]	delta [0.000001]	pipe cmd
0x50 [8502.764368]	delta [0.000002]	host trg
                   this is the xrun command first in the recover process.
0x60 [8502.764370]	delta [0.000002]	comp CEs
0x70 [8502.764371]	delta [0.000001]	value 0x0000000000000003
                   the command is xrun, but the current state is not ACTIVE but PREPARE.
                   from this point, it seems the previous xrun recover process has set the state
                   at PREPARE, but have not finished the next step, it was interrupted by 
                   the new xrun recover process.
                   the XRUN command comes and found the state is set to PREPARE already.
                   it will return with error.

lgirdwood, 2018-07-16:
please also add rmbox error log. just run "rmbox". How reproducible in % ?

tlauda, 2018-07-18:
Please verify with 4ba4aa8.

xiulipan, 2018-07-19:
after run aplay and arecord at same time
inject XRUN for ALSA
`echo 1 > /proc/asound/card0/pcm0p/sub0/xrun_injection`

[  336.761868] sof-audio sof-audio: error : DSP panic!
[  336.761886] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[  336.761894] sof-audio sof-audio: status: fw entered - code 00000005
[  336.761920] sof-audio sof-audio: error: runtime exception
[  336.761927] sof-audio sof-audio: error: trace point 00004000
[  336.761934] sof-audio sof-audio: error: DSP Firmware Oops
[  336.761942] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  336.761951] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x00060534 PS       0x00060025 SAR     0x00000020
[  336.761959] sof-audio sof-audio: EPC1     0xbe014732 EPC2     0xbe014724 EPC3     0xbe014724 EPC4    0x00000000
[  336.761967] sof-audio sof-audio: EPC5     0xbe01c28e EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  336.761999] sof-audio sof-audio: EPS2     0x00060720 EPS3     0x00060720 EPS4     0x00000000 EPS5    0x00060520
[  336.762007] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[  336.762090] sof-audio sof-audio: stack dump from 0xbe07fee0
[  336.762125] sof-audio sof-audio: 0xbe07fee0: 0xbe014732 0x00000000 0x00000020 0x00000000
[  336.762139] sof-audio sof-audio: 0xbe07fee4: 0x00060520 0x00000000 0x00060534 0x00000000
[  336.762153] sof-audio sof-audio: 0xbe07fee8: 0x0000000d 0x00000000 0xbe0237d9 0xbe0237df
[  336.762190] sof-audio sof-audio: 0xbe07feec: 0x00000000 0x00000000 0xbe0468c4 0x00000000
[  336.762248] sof-audio sof-audio: 0xbe07fef0: 0xbe01743e 0x00000000 0xbe0237d9 0x00000000
[  336.762267] sof-audio sof-audio: 0xbe07fef4: 0xbe012cb0 0x00000000 0x00060520 0x00000000
[  336.762288] sof-audio sof-audio: 0xbe07fef8: 0xe0040000 0x00000000 0x00000000 0x00000000
[  336.762307] sof-audio sof-audio: 0xbe07fefc: 0xbe011385 0x00000000 0x00000000 0x00000000
[  336.762327] sof-audio sof-audio: error: waking up any trace sleepers

After the dsp panic happen, the headset still play out noise.
Suspect that we have problem with dai DMA or SSP.

lgirdwood, 2018-07-19:
@xiulipan you have some memory dereference of 0x00060534 around the code at 0xbe014732, looks like it's crashed when running at level 1 (IPC idle level)

xiulipan, 2018-07-20:
@lgirdwood  this is very similar to issue #19 panic message.
How could you tell it
> crashed when running at level 1 (IPC idle level)
I found that the PS is modified by our panic_rewind function and always disable all interrupts.

lgirdwood, 2018-07-20:
@xiulipan it looks like level 5 actually, I misread PS. i.e level 1 -> 5 sequence and possibly the same corruption problem as https://github.com/thesofproject/sof/issues/19. Have you managed to do any binary chop of code yet to find the issue ?

xiulipan, 2018-07-23:
@lrgirdwood 
I checked the xtensa manual and found that the PS is actually be changed by the panic_rewind function. In that function we disable all interrupt by setting the PS.INTLEVEL to 5 to disable all interrupt level below and equal 5.
`         asm volatile("rsil      %0, 5"
                   : "=a" (flags) :: "memory");
` 
So I found it is hard to know which interrupt is coming as we got into panic.

> Have you managed to do any binary chop of code yet to find the issue ?

Find that simple remove will make status error, will done that with more modification.

keqiaozhang, 2018-08-02:
Confirmed  that this issue is fixed by PR#178.

closed 2018-08-02 by keqiaozhang