thesofproject/sof #102: 'ipc timed out' on GordonPeak once boot up

zhuqingliang, 2018-07-13:
environment:
sof-dev: a3a39f21218c074c71cd28b0c2c782b1626df3e3
sof:  2d92796072856e530c3a1bce4f0a9ce2c74692d4
soft: 47339d2302f445c3b35d50f9720ee649dd2f739b
tplg file name :
topology/sof-apl-tdf8532.tplg

reproduce steps:
check dmesg log once boot up

log:
[dmesg7.log](https://github.com/thesofproject/sof/files/2193135/dmesg7.log)
[rmbox7.log](https://github.com/thesofproject/sof/files/2193136/rmbox7.log)

tlauda, 2018-07-14:
Fixed on master. Please confirm and close.

keqiaozhang, 2018-07-16:
This issue is fixed.

closed 2018-07-16 by keqiaozhang

Jocelyn-Li, 2018-07-20:
@Jocelyn-Li test

Jocelyn-Li, 2018-07-20:
@Jocelyn-Li test

Jocelyn-Li, 2018-07-20:
@Xu Guangxin  test

Jocelyn-Li, 2018-07-20:
@Xu Guangxin  test

Jocelyn-Li, 2018-07-20:
@xuguangxin test

Jocelyn-Li, 2018-07-20:
@tlauda can you update the PR fix here?

xiulipan, 2018-07-24:
Fixed by https://github.com/thesofproject/sof/issues/99