thesofproject/sof #103: volume: zipper noise with fast changes

plbossart, 2018-07-13:
When I play on Up2 w/ the PCM512x codec, I can hear zipper noise when I keep pressing on up and down arrows to change the PGA Master Volume. This was reported before but apparently the fixes aren't enough. I view this as a bug, volume changes should not introduce audio artifacts (same as soft mute).

tlauda, 2018-07-13:
@plbossart HiFi3 or generic implementation?

plbossart, 2018-07-13:
This was on APL so I guess HiFi3.

lgirdwood, 2018-07-14:
@singalsu this may need ZC or a smaller step size.

singalsu, 2018-07-23:
@lgirdwood Sure, I will check some options after doing the updates to my UP2. Also as 3rd option logarithmic step could cope with larger step length. Zero crossings detection based volume update could be best but it would need largest amount of work.

singalsu, 2018-07-24:
I can now repeat the issue with my UP2+HiFiBerry after some struggle with fresh code. It's a bit hard to hear the zipper with music but with sine (997 Hz) it's quite audible.

singalsu, 2018-07-24:
The effect is also a bit different when pressing cursor keys in alsamixer vs. running successive commands like
'amixer sset 'PGA1.0 Master Playback Volume 1',0 10 10'
and
'amixer sset 'PGA1.0 Master Playback Volume 1',0 32 32'
so there may be two different issues. For the cursor keys case I wonder if the previous ramp has completed before the new target gain is received.

Note: I'm testing with gcc compiled FW. But it looks like the HiFi3 version just uses different copy function versions while the volwork code is shared so the issue should be the same.

singalsu, 2018-07-25:
I will submit a pull request with 250 ms long linear volume ramp (was 64 ms) and with gain value update rate increased to  every 1 ms (was 2 ms). I can still hear very silent zipper noise with sine wave but no more with music. If it's considered still too noisy maybe there's still room for some ramp length increase. Or consider changing ramp to logarithmic (still a small change) or even use ZC detection (major change, includes HiFi3 optimized code).

PR #148 (2018-07-25): Volume: Mitigate zipper noise by adjusting gain ramp parameters
Fixes: #103 "volume: zipper noise with fast changes".

The volume ramp length is increased from 64 ms to 250 ms. The vol_work
rate to update gain value is increased from every 2 ms to 1 ms. The
volume ramp macros are redefined to enter directly the desired ramp
length in milliseconds to ease the tuning.

Also the volume gain Qxy notation (currently Q2.16) is defined with macros
to avoid magic numbers in code in ramp computation.
commit 05a85741 (2018-07-25): Volume: Mitigate zipper noise by adjusting gain ramp parameters

Fixes: #103 "volume: zipper noise with fast changes".

The volume ramp length is increased from 64 ms to 250 ms. The vol_work
rate to update gain value is increased from every 2 ms to 1 ms. The
volume ramp macros are redefined to enter directly the desired ramp
length in milliseconds to ease the tuning.

Also the volume gain Qxy notation (currently Q2.16) is defined with macros
to avoid magic numbers in code in ramp computation.
closed 2018-07-25 by lgirdwood