thesofproject/sof #95: Pop noise: when pause/release during playing music

zhigang-wu, 2018-07-13:
When you pause/release during the playing music, you will hear the pop noise during this process.
It seems this pop noise is caused by lacking of the fade out fade in feature.
playing the silence area to music area, there is no volume ramp up process.
playing the music area to silence area, there is no volume roll down process.

zhigang-wu, 2018-07-13:
this can be use a sine wave (or constant data wave file) to confirm this.
get the result of I2S with loopback mode or result of amplifier's output with scope.

lgirdwood, 2018-07-14:
@zhigang-wu this is a feature request for a fader. Please change title to feature request and label appropriately.

jajanusz, 2019-06-06:
@singalsu @lgirdwood Are we going to make it? Or maybe we can assume that you can use volume ramp as fader?

singalsu, 2019-06-06:
Using volume would be idea since we have those in every pipeline. But does ALSA/ASoC have capability to do such control or would this need to be additional control layer within SOF?

singalsu, 2019-06-06:
Also maybe this isn't no more an issue with audio servers like CRAS and Pulseaudio?

lgirdwood, 2019-06-06:
@singalsu we can use std ALSA kcontrols to control faders in volume. They could be added to volume component topology as to whether a certain volume control exposed fader functionality..

kv2019i, 2019-06-06:
@singalsu I'm not convinced this should be done at the low-level. In most end-user systems audio server (Pulseaudio, CRAS, JACK) is producing constant audio stream to ALSA and music pause is not directly visible to the ALSA device. If a low-level app directly pauses, SOF will not be getting PCM samples, so it's a bit tricky from where you get the samples to fade out with. You'd have to add some dedicated delay (and latency) to always have some PCM data in the buffers to fade-out if needed.

tlauda, 2019-10-25:
@singalsu @kv2019i What is the idea here? Needed in FW or not?

aiChaoSONG, 2022-01-25:
close, because we already have volume component, and we have ramp up in there. No pop-noise found.

closed 2022-01-25 by aiChaoSONG