thesofproject/sof #96: fail to load all codec topologies under topology/test directory  on GP platform

zhuqingliang, 2018-07-13:


tlauda, 2018-07-13:
From dmesg3.log:
[    6.514771] sof-audio sof-audio: ASoC: physical link SSP4-Codec (id 0) not exist
[    6.514777] sof-audio sof-audio: error: tplg component load failed -22
[    6.514789] sof-audio sof-audio: error: failed to load DSP topology -22
[    6.514792] sof-audio sof-audio: ASoC: failed to probe component -22

It's drivers issue, not FW. Can you confirm and close?

xiulipan, 2018-07-16:
@zhuqingliang 
Please move this bug into SOFT repo.
https://github.com/thesofproject/soft/issues

This bug is because GP has multiple DAIs and test topology is all set with dai id to be 0

closed 2018-07-16 by xiulipan

Jocelyn-Li, 2018-07-20:
Moved to https://github.com/thesofproject/soft/issues/27