thesofproject/sof #98: arecord "input /output error"  using nocodec tplg on GordonPeak

zhuqingliang, 2018-07-13:
description:
aplay failed , but arecord  works well . The bug is probably related to 	https://github.com/thesofproject/sof/issues/97:'ipc timed out' in dmesg log when boot up using nocodec tplg on GordonPeak

environment:
sof-dev: a3a39f21218c074c71cd28b0c2c782b1626df3e3
sof: 2d92796
soft: 47339d2302f445c3b35d50f9720ee649dd2f739b
tplg file name :
topology/sof-apl-nocodec.tplg
topology/test/test-ssp5-mclk-0-I2S-volume-s16le-s16le-48k-24576k-nocodec.tplg
topology/test/test-ssp5-mclk-0-I2S-volume-s24le-s24le-48k-24576k-nocodec.tplg

reproduce steps:
aplay -Dhw:0,0 -c2 -f dat -r 48000 /dev/zero

log:
[dmesg1.log](https://github.com/thesofproject/sof/files/2192655/dmesg1.log)
[rmbox2.log](https://github.com/thesofproject/sof/files/2192657/rmbox2.log)

zhuqingliang, 2018-07-13:
the bug is same as sof bug , on c454399e8890c1d7eb007857f0dbe018d244eeb7 , the bug is gone.

closed 2018-07-13 by zhuqingliang

keqiaozhang, 2018-07-20:
This issue is fixed by PR#99.