thesofproject/sof #106: The volume of rear channels can not be changed on GP platform.

zhigang-wu, 2018-07-16:
when you play the four-channel music file on the GP platform, you will find the rear channel volume
can not be changed. the root cause is:
the tplg did not support this feature.
the "C_CONTROLMIXER" in "pipe-volume-playback.m4" and "pipe-volume-capture.m4"
only support two channels volume change.

I am not familiar with the m4 language. and who can help fix this issue?
but please pay attention during your fix, this change would affect the tplg script in the test folder.

tlauda, 2018-07-16:
Should be moved to soft repo.

lgirdwood, 2018-07-16:
@RanderWang please open this as topology issue.

closed 2018-07-16 by lgirdwood

Jocelyn-Li, 2018-07-20:
Moved to https://github.com/thesofproject/soft/issues/31