thesofproject/sof #107: noise issue when both front L and front R are connected on GP-MRB

keqiaozhang, 2018-07-16:
This issue occurs when both front L and front R are connected, but disconnect one of them, the noise will disappear.
sof-dev:a3a39f21218c074c71cd28b0c2c782b1626df3e3
sof-master:c4a9cca0a94969fbf1ecb42d660c0e40279d9ffa
soft-master:2abbfc0910ce9767d01462638d719309746844d1
tplg:sof-apl-tdf8532.tplg

how to reproduce:
1. ensure both front L and front R are connected.
2. aplay -D hw:0,0 -f S32_LE -r 48000 -c 4 /home/rander/4ch_32bit.wav , you can hear the noise and the noise sounds like a rustle.
3. disconnect front L or front R. the noise will disappear

lgirdwood, 2018-07-16:
This does not appear to be a firmware bug, perhaps the codec settings need adjusted to driver both L & R outputs here ?

keqiaozhang, 2018-07-16:
No codec kctls are exposed in amixer setting.
[image]

lgirdwood, 2018-07-16:
The codec driver then. Should be logged as a driver issue.

keyonjie, 2018-07-16:
tdf8532 codec driver doesn't export any kcontrol at the moment, it's controlled via ioc i2c command, if we can make sure the data is correct on SSP pins, better to report them to Dirana or integration team.

zhigang-wu, 2018-07-16:
no sure this noise is still exist during playing the sine wave or not, if yes we could use the scope to probe the point of analog output.

plbossart, 2018-07-16:
> tdf8532 codec driver doesn't export any kcontrol at the moment, it's 
> controlled via ioc i2c command, if we can make sure the data is correct 
> on SSP pins, better to report them to Dirana or integration team.

Yes, there is no requirement for volume control in the firmware for that 
specific board but still it's a good issue to fix to make sure we can do 
volume control for 2+ channels

keqiaozhang, 2018-07-16:
This issue is moved to thesofproject/linux#34

closed 2018-07-16 by keqiaozhang