thesofproject/sof #115: Stress Test: long time simultaneous playback and capture will cause IPC timed out on UP^2

zhuqingliang, 2018-07-17:
description:
aplay and arecord both run . After 20 minutes more or less , both stopped and prompt ' Input/output error',
then check dmesg log , sometimes there will be 'ipc timed out'.

System Environment
sof-v4.14 :  cae4773bbd83b392be18c7d6fc1b35369b7ed715
sof:  c4a9cca0a94969fbf1ecb42d660c0e40279d9ffa
soft:2abbfc0910ce9767d01462638d719309746844d1

reproduce steps:
aplay -Dhw:0,0 -fdat -c2 -r 48000 /dev/zero & arecord -Dhw:0,0 -fdat -r 48000 -c2 test.wav
dmesg | grep error

log:
[dmesg.log](https://github.com/thesofproject/sof/files/2200333/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2200334/rmbox.log)

lgirdwood, 2018-07-17:
Related to XRUN handler running. Probably a duplicate of XRUN bug ?

tlauda, 2018-07-18:
Please verify with fd55d5f7d869e3fef92f579fcce19b52f6aa4f57.

tlauda, 2018-07-24:
@xiulipan 
Capture xrun should be fixed on top of c65d73397cc04da4600e02b7d91ac0423dc55ece.

zhuqingliang, 2018-07-24:
I test with  7a2f8c20777caac678b53810da37875bea3ac2fe  for 30 minutes , the issues do not reproduce. The bug is gone.

closed 2018-07-24 by zhuqingliang

tlauda, 2018-07-24:
Fixed by #143.