thesofproject/sof #116: PCM read/write error when buffer size < 768 on APL

keqiaozhang, 2018-07-17:
Today I tried to test playback with specified buffer size and found that if the buffer size is less than 768, it will easy introduce I/O error  and even with IPC timed out. 
How to reproduce:
1. aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/zero --buffer-size=384
  It will stop with pcm read/write error.
[dmesg-bufsize.log](https://github.com/thesofproject/sof/files/2200760/dmesg-bufsize.log)
[rmbox-buffsize.log](https://github.com/thesofproject/sof/files/2200763/rmbox-buffsize.log)

lgirdwood, 2018-07-17:
This looks like another duplicate of the XRUN bug. @keqiaozhang can you check.

plbossart, 2018-07-17:
Don't we have by-design limitations on the buffer/period sizes? If the topology states that the host DMA can work with 48 samples, that's 192 bytes and already larger than the period in the firmware you are using in this case on the host side. I'd think the ALSA period needs to be >= the firmware period?

keqiaozhang, 2018-07-18:
I will check this tomorrow.

lgirdwood, 2018-07-19:
@keqiaozhang ok, please close once you are satisfied.

closed 2018-07-20 by keqiaozhang

keqiaozhang, 2018-07-20:
This issue is fixed by PR#125.

Jocelyn-Li, 2018-07-20:
This issue is fixed by https://github.com/thesofproject/sof/pull/125