thesofproject/sof #122: Stress Test: 'DSP panic' after long time simultaneous playback and capture on BYT

zhuqingliang, 2018-07-18:
description:
aplay and arecord both run . After 1 to 3 minutes , both stopped and prompt ' Input/output error',
then check dmesg log ,  there will be 'DSP panic' ：
[  240.561830] sof-audio sof-audio: error : DSP panic!
[  240.561861] sof-audio sof-audio: panic: dsp_oops_offset 1329152 offset 1329152
[  240.561899] sof-audio sof-audio: error: can't enter idle
[  240.561917] sof-audio sof-audio: error: trace point 60040000
[  240.561932] sof-audio sof-audio: error: DSP Firmware Oops
[  240.561947] sof-audio sof-audio: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[  240.561967] sof-audio sof-audio: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00060525 SAR     0x0000001f
[  240.561985] sof-audio sof-audio: EPC1     0xff2c17bd EPC2     0xff2cae01 EPC3     0xff2cae01 EPC4    0xff2c0d66
[  240.562003] sof-audio sof-audio: EPC5     0xff2cae01 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  240.562021] sof-audio sof-audio: EPS2     0x00060320 EPS3     0x00060320 EPS4     0x00060520 EPS5    0x00060320
[  240.562039] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00076f48 INTERRU 0x000000a2
[  240.562055] sof-audio sof-audio: stack dump from 0xff327e60
[  240.562074] sof-audio sof-audio: 0xff327e60: 0x00000002 0x00000000 0x00009800 0x30000000
[  240.562093] sof-audio sof-audio: 0xff327e64: 0xff2ce8e4 0xff327ea0 0xff301d28 0xff3400c0
[  240.562112] sof-audio sof-audio: 0xff327e68: 0xff2c1d00 0xff306960 0xff30699c 0xff30694c
[  240.562130] sof-audio sof-audio: 0xff327e6c: 0x7f2cae01 0xff327ec0 0x000000ff 0x00000000
[  240.562149] sof-audio sof-audio: 0xff327e70: 0x00000008 0x000002c0 0x000002c8 0x00000000
[  240.562168] sof-audio sof-audio: 0xff327e74: 0xbf2c0d2b 0xff327fc0 0x0124f800 0xff2cdd00
[  240.562187] sof-audio sof-audio: 0xff327e78: 0x00000001 0x00000000 0x0000001e 0x00000000
[  240.562205] sof-audio sof-audio: 0xff327e7c: 0xff302914 0xff327f70 0x00000007 0x00000088
[  240.562221] sof-audio sof-audio: error: waking up any trace sleepers

System Environment:
sof-dev:  26c56d760e2a05cb4d30e7453a82493e6582133c
sof:         e69ebddb4f61a0e7047efb728f7d7525171b6690
soft:        576d15b9b48e91147d9654ea894e5971812f0550
using :  test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

log:
[dmesg.log](https://github.com/thesofproject/sof/files/2204877/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2204878/rmbox.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2204879/rmbox-t.log)

tlauda, 2018-07-18:
Please verify with fd55d5f7d869e3fef92f579fcce19b52f6aa4f57.

zhuqingliang, 2018-07-20:
I test with  d696d8f8bad1490575a8293e0982786300eb52db , the bug is still exist.

lgirdwood, 2018-07-20:
This oops message means we have taken too long to enter the idle state, i.e. something is preventing the CPU from enter WAITI state. This could be a deadlock, forever loop or some code is not yeilding the CPU.

zhuqingliang, 2018-07-23:
The bug still exist on 7a2f8c20777caac678b53810da37875bea3ac2fe

lgirdwood, 2018-07-26:
@zhigang-wu please lookup each EPC register when you next reproduce from the dissasembly and log the  functions that each EPC is running in.

RanderWang, 2018-07-27:
It is caused by XRUN.  you would get kernel panic when XRUN happens.  I also checked the EPC register and didn't find any useful info. 

I tried the https://github.com/thesofproject/sof/pull/129/commits/390e60f82266b80e1256ffacf0dd6e6a4b3ca2dd suggested by zhigang.  This patch kills the panic

keqiaozhang, 2018-08-02:
This issue cannot be reproduced after more than 20mins. So close this bug.

closed 2018-08-02 by keqiaozhang