thesofproject/sof #128: Stress test: noise issue when keep injecting xrun

keqiaozhang, 2018-07-19:
I found this issue on up^2 with pcm512 tolg. when I playing a 48K_32bits wav file and keep injecting xrun, then I can hear noises from time to time. 

How to reproduce this issue:
1. aplay -D hw:0,0 -f S32_LE -r 48000 -c 2 48K_32bits.wav
2. inject xrun in other terminal.

How to inject run:
You can inject xrun by writing to a proc file when CONFIG_SND_PCM_XRUN_DEBUG is set in kernel config.
For this case, please run: echo 0 > /proc/asound/card0/pcm0p/sub0/xrun_injection,
I wrote a script for it, please refer:
root@test-UP:/home/test# cat xrun_injection.sh
#!/bin/bash

int=0
while [ $int -lt 100000 ]
do
        echo 0 > /proc/asound/card0/pcm0p/sub0/xrun_injection
        sleep 1
        let int++
done

[dmesg-xrun.log](https://github.com/thesofproject/sof/files/2208996/dmesg-xrun.log)
[rmbox-xrun.log](https://github.com/thesofproject/sof/files/2208997/rmbox-xrun.log)

lgirdwood, 2018-07-19:
The dmesg attached is the wrong log. Can you also attach audacity trace.

keqiaozhang, 2018-07-19:
@plbossart 
The noise I reported here is not the pop issues, I'm pretty sure about this. I tested with hifiberry codec, I will find another way to record the data. This noise come and go during the test and Xiuli also reproduce this issue. 
@xiulipan 
Please help to give more tech explanation.

xiulipan, 2018-07-20:
@plbossart maybe @keqiaozhang test is too aggressive, actually you can random inject xrun just like pasue/resume the pipeline. Then you will hear distortion noise just like the UP2 one.

lgirdwood, 2018-07-20:
@keqiaozhang , @xiulipan please be more descriptive about the noise/distortion. i.e how long does it last, does the audio recover. Please also attach a audacity screenshot.

xiulipan, 2018-07-23:
@lgirdwood will update the loopback PCM audacity screenshot.

This distortion is similar to the pause/release noise.
> how long does it last

It will last to end if we did not do anything after we heard the noise
> does the audio recover

The audio may recover after another XRUN injection (thus a stop/start pattern)

tlauda, 2018-07-23:
@xiulipan 
Please verify with if issue still exists with 956585e089d4c9d43621575108bda4d7905eb99d.

keqiaozhang, 2018-07-23:
@tlauda 
Verified with the latest commit 7a2f8c20, the issue still exists. The audacity screenshot and the audio wav are attached.
I tested it with hifiberry codec and record the data via another machine. 
Test steps:
1. Play a 48k-32bit-997Hz- sine wav
2. Inject the xrun per second.
3. Capture the data.
Noise occurs and disappears at random when injecting xrun.
[image]
[noise.zip](https://github.com/thesofproject/sof/files/2218598/noise.zip)

tlauda, 2018-07-24:
@keqiaozhang @xiulipan 
With all my patches I've verified that HDA DMA and GPDMA pointers are correct after stop and start sequence and they don't overlap each other. Transfers should be continued correctly without any glitches, but for some reason 1 in 5 times the stream is corrupted. The only last thing I can think of is some kind of data inconistency on kernel side. I've also noticed that kernel clears the RUN bit before sending IPC to FW, which seems to be invalid.

plbossart, 2018-07-24:
@tlauda what is the expected behavior then? I thought we were done with the HDAudio DMA programming sequence issues, looks like it's still wrong?

xiulipan, 2018-07-25:
Test with L300Hz R400Hz Sine wave on UP2 with hard link loopback on SSP5 with XRUN injection.

Find some strange thing after STOP/START ( the whole test did not get any noise issues).
After start a few samples ( 100 samples) are following the STOP moment, then it continue on with some other data that can not easy find belong to.

[image]

tlauda, 2018-07-25:
@plbossart 
Proposed flow was supposed to be the three step procedure:
1. Send IPC (maybe PAUSE?) to FW to stop updating read/write pointer of Host DMA.
2. Set RUN = 0.
3. Send IPC (STOP?) to FW to disable Host DMA gateway.

Currently we are doing steps 1 & 2 after receiving COMP_TRIGGER_STOP IPC, which is after driver sets RUN = 0. So as you can see, FW for a moment will still read/write Host DMA buffer, which will not only cause some kind of data inconstistency, but also can cause the hang of the DMA.
Since Host DMA refactor in FW is still work in progress, I would propose to change the flow in driver and set RUN = 0 after sending COMP_TRIGGER_STOP IPC to FW. Also driver should poll this bit after setting just to make sure, that the stream has finished start/stop procedure. I think this part of code is missing.

lgirdwood, 2018-07-26:
@tlauda, @plbossart , @xiulipan we zero a period during xrun iirc. It looks like we may need to either fill period with last value or fade/unfade. @singalsu ?

tlauda, 2018-07-26:
@lgirdwood 
In this situation there is no xrun in FW, only in ALSA.
From FW perspective it is stop - start sequence, during which buffers aren't cleared.

lgirdwood, 2018-07-26:
@tlauda ok, so a fade is something @singalsu may look at in the future.

tlauda, 2018-07-26:
I think we should try reset the stream after the stop sequence and reinitialize it before the start. That would also require to prepare() pipeline before the start to reinit the buffers. This way we're going to have the clean slate and basically the same situation as before the first start. There shouldn't be any noise.

zhigang-wu, 2018-07-26:
@tlauda 
I found something, it seems there is mismatch between the host dma's hardware read/write pointer and the buffer's read/write pointer sometimes. 
I think it is a hardware issue. this mismatch happen sometimes. I think that is the reason: when i use
the prepare() patch, it still has the noise issue.
I attached two files: "0001-test-add-the-trace-code-to-check-hw-pointer-mismatch.txt" is the test code i used to check the mismatch status. the "log-0725.txt" is the log.

the host dma's hardware read pointer and write pointer are different value after enable the bit in DGCS
register. that is correct. so from now on, the hardware read pointer and write pointer should always different.
but from the log, you can see, sometimes they became same.
you can check the trace code i used to check the print location to confirm what i found.
you can go through the log file i attached already.

I get two piece of log over here to analysis:

this piece of log is the wrong status, the buffer write pointer and the host dma write pointer are
mismatch. the host dma write pointer should be 0x300.
0x220 [166.973390]	delta [0.000039]	host upd
0x230 [166.973393]	delta [0.000002]	buffer pro
0x240 [166.973394]	delta [0.000001]	value 0x00000000be001380   --- buffer read pointer
0x250 [166.973395]	delta [0.000001]	value 0x00000000be001380   --- buffer write pointer
0x260 [166.973396]	delta [0.000001]	value 0x00000000be001080   --- buffer base address
0x270 [166.973398]	delta [0.000002]	value 0x0000000000000000   --- host dma read pointer
0x280 [166.973403]	delta [0.000004]	value 0x0000000000000000   --- host dma write pointer

this piece of log seems correct.
0x8c20 [167.069390]	delta [0.000039]	host upd
0x8c30 [167.069392]	delta [0.000002]	buffer pro
0x8c40 [167.069394]	delta [0.000001]	value 0x00000000be001380  --- buffer read pointer
0x8c50 [167.069395]	delta [0.000001]	value 0x00000000be001380  --- buffer write pointer
0x8c60 [167.069396]	delta [0.000001]	value 0x00000000be001080  --- buffer base address
0x8c70 [167.069398]	delta [0.000002]	value 0x0000000000000000  --- host dma read pointer
0x8c80 [167.069399]	delta [0.000001]	value 0x0000000000000300  --- host dma write pointer

[0001-test-add-the-trace-code-to-check-hw-pointer-mismatch.txt](https://github.com/thesofproject/sof/files/2231372/0001-test-add-the-trace-code-to-check-hw-pointer-mismatch.txt)
[log-0725.txt](https://github.com/thesofproject/sof/files/2231373/log-0725.txt)

tlauda, 2018-07-26:
@zhigang-wu 
It's more likely that write pointer has wrapped around to 0. You should additionaly read DGCS and check BF bit.

lgirdwood, 2018-07-26:
@tlauda rather than sending two additional IPCs for reset() + prepare() after STOP | PAUSE, we should probably do this automatically in FW after STOP | PAUSE. That way we have lower latency to restart/recover a stream since we have two less IPC messages ? 
@zhigang-wu can you have a look at calling reset() and prepare() in firmware after STOP | PAUSE and evaluate.

zhigang-wu, 2018-07-26:
@lgirdwood @tlauda 
I checked with another test code, i will attached the test code and log after i am back to office. 
from that log, it seems no matter it is the first start or start after xrun (no matter how many xrun happen). the host dma's hw pointer will show the right value: r_ptr=0x00, w_ptr=0x300. 
let me explain it.
1. this is the first start, the host dma pointer will change like below:
the initial value is : r_ptr=0x0, w_ptr=0x0;  
the first normal start (enable dma by set DGCS), the value will be: r_ptr=0x0, w_ptr=0x300
after pipeline schedule, the dma_copy trigger, the value will be: r_ptr=0x300, r_ptr=0x0
2.  this is the start after the xrun happen, the host dma pointer will change like below:
the xrun start (enable dma by set DGCS), the value never changes, it is always: r_ptr=0x0, w_ptr=0x300, I can confirm this by the log.
after pipeline schedule, the dma_copy trigger, the value will be: r_ptr=0x300, r_ptr=0x0. because the log will miss, i did not confirm this point.

I suspect this mismatch happen after the dma_copy() (this is host dma's trigger in the pipeline). 
the dma_copy() sometimes did not trigger the w_ptr value to change.
but this kind of mismatch issue only happen when the ALSA xrun happen.

another point we have to pay attention, this mismatch is not always happen.
it has some possibility.

@lgirdwood 
I will do the reset and prepare after the stop/pause command tomorrow.

lgirdwood, 2018-07-26:
@tlauda  great thanks :)

zhigang-wu, 2018-07-26:
It's more likely that the write pointer has wrapped around to 0. You should additionaly read DGCS and check BF bit.
---------------------------------------------------------------------
@tlauda 
you means before the dma_copy() is called, we should check the BF bit in the DGCS, right?  I will try this.
maybe before the dma_copy() is called, the w_ptr is changed already. something abnormal happened.

lgirdwood, 2018-07-26:
@zhigang-wu looks like tlauda will fix this tomorrow, can you look at the other outstanding issues.

zhigang-wu, 2018-07-27:
@lgirdwood 
OK. I will move on other issue.

xiulipan, 2018-07-30:
maybe fixed by #170 
please have a try.

keqiaozhang, 2018-07-31:
@xiulipan 
I cannot reproduce this issue on BYT. 
On APL, this issue still exists, but the reproduce rate is reduced to <20%.

keqiaozhang, 2018-08-02:
Update:
This issue still can be reproduced on BYT APL and CNL and the reproducibility is low about 10%.

keqiaozhang, 2018-08-02:
Confirmed that this issue is fixed by PR#170 on BYT APL and CNL.

closed 2018-08-02 by keqiaozhang