thesofproject/sof #152: DMIC: 16bits FIFOs record at double rate

jajanusz, 2018-07-26:
DMIC works correctly only for 32bit (fifo_bits_a), for 16b it records at double rate - for example 10s squeezed into 5s with 2x tempo.

lgirdwood, 2018-07-26:
@singalsu looks like DMA width is wrong for 16bit mode. probably reading back 32bits.

singalsu, 2018-07-26:
I will do checks once I'm able to get this working again UP2 (code in dai.c line 671->) . I have still some issue with topology and kernel.

singalsu, 2018-07-26:
I can repeat the bug now. With DMIC the 16 bit stereo data frame is packed to 32 bit word as <L><R>. And mono data is packed as <L><L>. Is SSP similar? If SSP packs 16 bit data differently it could explain this.

singalsu, 2018-07-26:
Testing this
```
dd->config.src_width = 4;
```

ranj063, 2018-07-26:
@singalsu I think this is an error. I've sent you a patch that addresses this issue by setting the config only for the comp who's dai_index matches the dai_index in the config. Please give it a try.

singalsu, 2018-07-27:
I noticed that  if I set "dd->config.burst_elems = 4;" for DMIC in dai_config() the change is seen in dw-dma.c. Therefore it is very likely that some later code overwrites dd->config.src_width to 2. Maybe some SSP specific thing kicks in?

singalsu, 2018-07-27:
I can make DMIC 16bit capture work by commenting out these lines from dai_capture_params()

```
	/* set up DMA configuration */
	config->direction = DMA_DIR_DEV_TO_MEM;
	//config->src_width = comp_sample_bytes(dev);
	//config->dest_width = comp_sample_bytes(dev);
	config->cyclic = 1;
	config->src_dev = dd->dai->plat_data.fifo[1].handshake;
```
My question is that can this SSP specific thing be done instead in dai_config for type SOF_DAI_INTEL_SSP?

jajanusz, 2018-07-27:
@singalsu 
I think we shouldn't set config fields in many places, it will get messy, better to set config once in *_params functions and then use them in dai_config as we do now. We have to find better way to figure out what widths to set before passing config to dai_config

singalsu, 2018-07-27:
Then there should be DAI type check in _*params and apply differently src_width and dest_width.

singalsu, 2018-07-27:
I'll make a pull request with edits to dai_capture_params().

plbossart, 2018-08-08:
@singalsu Looks like this issue was solved, if yes let's close it?

jajanusz, 2018-08-08:
@plbossart 
@singalsu wrote how to fix it, but PR needs changes, we shouldn't set widths in set_config, so I'll set them in some proper callback while doing ssp mux/demux, it's not high priority issue, but please keep it open

jajanusz, 2018-11-16:
Fixed in #587

closed 2018-11-16 by jajanusz

mengdonglin, 2018-11-19:
@singalsu Would you please check if this issue can be reproduced on your APL?

jajanusz, 2018-11-19:
I checked it with my PR from friday that is already merged - It works. Is it still bugged on your apl with newest master?

singalsu, 2018-11-19:
@mengdonglin Yep, confirmed here too with UP2 that 2ch and 4ch 16 bit capture worked OK.