thesofproject/sof #153: cmocka ut: Unit tests should be uniquely identifiable by function names

jajanusz, 2018-07-26:
We have to use naming like we were using with first unit test (f.e. list, audio buffer) - name of the testing function should go like test_<path-to-header>_<condition>, like test_math_numbers_gcd_when_something_then_something, otherwise we will have in TAP logs on CIs logs like:
For files:
```
someheaderA\a.c
someheaderB\a.c
```
```
PASS: a - a_test_condition
FAIL: a - a_test_condition
```
And we don't know for which one it failed.
So please make your test names more verbose.

jajanusz, 2018-07-26:
@lgirdwood I asked them already to fix names. You can assign me to all UT patches.

slawblauciak, 2018-07-26:
Fixed in https://github.com/thesofproject/sof/pull/155

closed 2018-07-26 by jajanusz