thesofproject/sof #157: Trace: The trace output is not displayed when it arrives

singalsu, 2018-07-26:
The trace in APL platform doesn't update the display per trace event but in larger blocks only when more trace data arrives. E.g. it is not possible to see volume trace events for latest alsamixer key press for PGA gains. Or as other example I can't see DMIC capture trace events before I run another capture.

This makes debugging harder when the needed information need to be looked back from much earlier trace data.

lgirdwood, 2018-07-30:
Trace should be sent every 500ms or when buffer > half full, whichever happens first. @RanderWang can you check.

RanderWang, 2018-08-07:
two issue for this case. I will create two PRs for this issue

xiulipan, 2018-08-13:
fixed by #204

keqiaozhang, 2018-08-14:
Confirmed this issue is fixed by thesofproject/sof/pull/204。

closed 2018-08-14 by keqiaozhang