thesofproject/sof #161: noise occurs intermittently when injecting xrun for playback on BYT

keqiaozhang, 2018-07-27:
This issue happens when injecting xrun and the noise issue interval occurs. For example, this time is good and next time will be bad.
how to reproduce:
1. play a wav file 
2. inject the xrun every 2 seconds
    echo 1  > /proc/asound/card0/pcm0p/sub0/xrun_injection
then you can hear the noise form time to time.
[dmesg-xrun.log](https://github.com/thesofproject/sof/files/2234959/dmesg-xrun.log)
[xrun-rmbox-t.log](https://github.com/thesofproject/sof/files/2234961/xrun-rmbox-t.log)

zhigang-wu, 2018-07-27:
this can be fixed with the prepare() operation before start.
I tested it on BYT already.

xiulipan, 2018-07-30:
fix by #169 and #170

keqiaozhang, 2018-07-31:
Confirmed this issue is fixed.

keqiaozhang, 2018-08-01:
We did a stress test on BYT today and found that this issue still can be reproduced, repetition is low about 5%.
But with PR #170, this issue is hard to reproduce. So let's reopen this bug until PR#170 is merged.

PR #170 (2018-08-02): host: remove reset in stop to keep status
Need to keep status of host comp at when we stop.

Check ALSA handler for these XRUN patterns to make sure no more bugs happen

depend on #169 to fix #161
closed 2018-08-02 by lgirdwood