thesofproject/sof #172: 'error log' in dmesg when boot up on CNL using sof-cnl-rt274.tplg

zhuqingliang, 2018-07-30:
description:
run dmesg once boot up , but aplay and arecord work well.

[    5.645901] cnl_rt274 cnl_rt274: snd-soc-dummy-dai <-> Passthrough 0 mapping ok
[    5.645962] cnl_rt274 cnl_rt274: ASoC: no sink widget found for Dummy Playback
[    5.645965] cnl_rt274 cnl_rt274: ASoC: Failed to add route ssp2 Tx -> direct -> Dummy Playback
[    5.645968] cnl_rt274 cnl_rt274: ASoC: no source widget found for ssp2_out
[    5.645970] cnl_rt274 cnl_rt274: ASoC: Failed to add route ssp2_out -> direct -> ssp2 Tx
[    5.645974] cnl_rt274 cnl_rt274: ASoC: no source widget found for Dummy Capture
[    5.645977] cnl_rt274 cnl_rt274: ASoC: Failed to add route Dummy Capture -> direct -> ssp2 Rx
[    5.645980] cnl_rt274 cnl_rt274: ASoC: no sink widget found for ssp2_in
[    5.645982] cnl_rt274 cnl_rt274: ASoC: Failed to add route ssp2 Rx -> direct -> ssp2_in
[    5.645986] cnl_rt274 cnl_rt274: ASoC: no sink widget found for Dummy Playback
[    5.645989] cnl_rt274 cnl_rt274: ASoC: Failed to add route ssp1 Tx -> direct -> Dummy Playback
[    5.645993] cnl_rt274 cnl_rt274: ASoC: no source widget found for ssp1_out
[    5.645996] cnl_rt274 cnl_rt274: ASoC: Failed to add route ssp1_out -> direct -> ssp1 Tx
[    5.646797] input: sof-cnl-audio Headset as /devices/pci0000:00/0000:00:1f.3/cnl_rt274/sound/card0/input8

System Environment:
sof-dev: b78ead1e0cd31113bc1b3b6d122c9afc1a344147 
sof: a2cfd8c5a03da54189d7cb650c77c1ae0cedfdc3
soft: debaf8f889607406159a13f8bba7d7218243683d 
using topology/sof-cnl-rt274.tplg

reproduce steps:
dmesg

log:
[dmesg.log](https://github.com/thesofproject/sof/files/2240965/dmesg.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2240966/rmbox-t.log)

tlauda, 2018-07-31:
I think it's soft/kernel issue.
Can we move this to the appropriate project?

zhuqingliang, 2018-07-31:
move to https://github.com/thesofproject/soft/issues/46

closed 2018-07-31 by zhuqingliang