thesofproject/sof #173: stress test for start playback and capture at sametime cause dma trace error on BYT

xiulipan, 2018-07-31:
reproduce step:
start playback
start and stop capture when error happen

rmbox log:

0x0 [188.520303]        delta [188.520303]      dma eS0
0x10 [188.520305]       delta [0.000003]        value 0x0000000000000007
0x20 [188.520308]       delta [0.000003]        value 0x0000000000000203
0x30 [188.520310]       delta [0.000002]        value 0x0000000000000003
0x40 [188.520313]       delta [0.000003]        buffer ebb
0x50 [188.520315]       delta [0.000002]        value 0x00000000fffffff0
0x60 [189.015096]       delta [0.494781]        dma eS0
0x70 [189.015098]       delta [0.000003]        value 0x0000000000000007
0x80 [189.015101]       delta [0.000003]        value 0x0000000000000203
0x90 [189.015103]       delta [0.000002]        value 0x0000000000000003
0xa0 [189.015106]       delta [0.000003]        buffer ebb
0xb0 [189.015108]       delta [0.000002]        value 0x00000000fffffff0
0x0 [188.520303]        delta [188.520303]      dma eS0
0x10 [188.520305]       delta [0.000003]        value 0x0000000000000007
0x20 [188.520308]       delta [0.000003]        value 0x0000000000000203
0x30 [188.520310]       delta [0.000002]        value 0x0000000000000003
0x40 [188.520313]       delta [0.000003]        buffer ebb
0x50 [188.520315]       delta [0.000002]        value 0x00000000fffffff0
0x60 [189.015096]       delta [0.494781]        dma eS0
0x70 [189.015098]       delta [0.000003]        value 0x0000000000000007
0x80 [189.015101]       delta [0.000003]        value 0x0000000000000203
0x90 [189.015103]       delta [0.000002]        value 0x0000000000000003
0xa0 [189.015106]       delta [0.000003]        buffer ebb
0xb0 [189.015108]       delta [0.000002]        value 0x00000000fffffff0

related code
src/drivers/dw-dma.c
```
	/* is channel idle, disabled and ready ? */
	if (p->chan[channel].status != COMP_STATE_PREPARE ||
		(dw_read(dma, DW_DMA_CHAN_EN) & (0x1 << channel))) {
		ret = -EBUSY;
		trace_dma_error("eS0");
		trace_error_value(dw_read(dma, DW_DMA_CHAN_EN));
		trace_error_value(dw_read(dma, DW_CFG_LOW(channel)));
		trace_error_value(p->chan[channel].status);
		goto out;
	}
```
src/lib/dma-trace.c
```
	/* copy this section to host */
	size = dma_copy_to_host_nowait(&d->dc, config, d->host_offset,
		buffer->r_ptr, size);
	if (size < 0) {
		trace_buffer_error("ebb");
		trace_error_value(size);
		goto out;
	}
```

xiulipan, 2018-07-31:
reproduce rate < 5 %

xiulipan, 2018-07-31:
when error happen, the DMA trace seems did not stop and continue try to call dma-copy but always get fail, the error trace will have upate all time.
rmbot log:
keep try to dma-copy every scheduler.
0x1b0 [517.302752]      delta [0.494783]        dma eS0
0x1c0 [517.302755]      delta [0.000003]        value 0x000000000000000f
0x1d0 [517.302758]      delta [0.000003]        value 0x0000000000000003
0x1e0 [517.302760]      delta [0.000002]        value 0x0000000000000003
0x1f0 [517.302763]      delta [0.000003]        buffer ebb
0x200 [517.797545]      delta [0.494783]        dma eS0
0x210 [517.797548]      delta [0.000003]        value 0x000000000000000f
0x220 [517.797551]      delta [0.000003]        value 0x0000000000000003
0x230 [517.797553]      delta [0.000002]        value 0x0000000000000003
0x240 [517.797556]      delta [0.000003]        buffer ebb
0x250 [518.292338]      delta [0.494783]        dma eS0
0x260 [518.292341]      delta [0.000003]        value 0x000000000000000f
0x270 [518.292344]      delta [0.000003]        value 0x0000000000000003
0x280 [518.292346]      delta [0.000002]        value 0x0000000000000003
0x290 [518.292349]      delta [0.000003]        buffer ebb
0x2a0 [518.787131]      delta [0.494783]        dma eS0

We may need some error handler for this case.

lgirdwood, 2018-08-01:
@xiulipan looks like DMA was stopped prior to DMA copy completion. This happens when the DMA is waiting on SSP FIFO handshakes to empty the DMA buffer. i.e. the SSP port is being disabled before DMA has finished copying period.

xiulipan, 2018-08-02:
@lgirdwood 
Thanks for the hint. I will check if we have some stop order issues there.
It seems on BYT all 3 DMA ( and maybe 4 DMA in the second log?)
The DMA TRACE did not have a FIFO connected but it still got into the enabled status.
I will double check if we have some error when handling the DMA stop/start if we failed one of them.

keqiaozhang, 2018-08-14:
This issue is hard to reproduce with latest code base, it may benefit from /thesofproject/sof/pull/189 and /thesofproject/sof/pull/208. Close this bug.

closed 2018-08-14 by keqiaozhang