thesofproject/sof #174: Stress test: start playback and capture at same time then inject xrun for playback and pause/release for capture will cause DSP panic

keqiaozhang, 2018-07-31:
Reproduce steps:
1. open a console and run: 
   aplay -D hw:0,0 -f dat -c 2 /dev/zero & arecord -D hw:0,0 -f dat -c 2 /dev/null -i 
2. Press the space key to do pause/release for capture.
3. Open another console and inject xrun for playback:
    echo 0 > /proc/asound/card0/pcm0p/sub0/xrun_injection

[dmesg.log](https://github.com/thesofproject/sof/files/2243981/dmesg.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2243982/rmbox-t.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2243983/rmbox.log)

tlauda, 2018-07-31:
Isn't it the duplicate of #168?

keqiaozhang, 2018-07-31:
@tlauda
Thanks for your input. I will try to revert PR143 and test it again,  if it cannot be reproduced, I will close this bug.

keqiaozhang, 2018-08-02:
Confirmed that this issue is fixed by PR#178.

closed 2018-08-02 by keqiaozhang