thesofproject/sof #175: Stress test: inject xrun and do pause/release for capture will cause DSP panic on GP-MRB

keqiaozhang, 2018-07-31:
How to reproduce:
1. run: arecord -Dhw:0,0 -f dat -c 2 /dev/null -i 
2. inject xrun for capture.
    echo 1 > /proc/asound/card0/pcm0c/sub0/xrun_injection
3. press space key to do pause/release for capture.

tplg: test-ssp4-mclk-0-I2S-volume-s16le-s16le-48k-24576k-nocodec.tplg

[rmbox.log](https://github.com/thesofproject/sof/files/2244189/rmbox.log)
[dmesg.log](https://github.com/thesofproject/sof/files/2244190/dmesg.log)

tlauda, 2018-07-31:
Isn't it the duplicate of #168?

xiulipan, 2018-07-31:
@tlauda I think they are the same

keqiaozhang, 2018-07-31:
@tlauda @xiulipan 
Thanks for your input. I will try to revert PR143 and test it again,  if it cannot be reproduced, I will close this bug.

keqiaozhang, 2018-08-02:
Confirmed that this issue is fixed by PR#178.

closed 2018-08-02 by keqiaozhang