thesofproject/sof #182: Adjust PGA volume control has no effect on GP-MRB

keqiaozhang, 2018-08-01:
How to reproduce:
1. aplay -D hw:0,0 -f S32_LE -r 48000 -c 4 4ch_32bit.wav
2. adjust the PAG value in alsamixer
The sound is keep loud, even set the volume as 0.

[volume_dmesg.log](https://github.com/thesofproject/sof/files/2248268/volume_dmesg.log)
[volume_rmbox.log](https://github.com/thesofproject/sof/files/2248269/volume_rmbox.log)

lgirdwood, 2018-08-02:
@keqiaozhang can you tell us which volume control you are changing.

keqiaozhang, 2018-08-02:
@lgirdwood 
I changed the PGA1.0 Master Playback Volume 1 when playback(I've basically tried all of PGA*), I retested this issue today, but unable to reproduce it, checked with Zhigang, it's more like a random issue, I will do more tests to confirm it.

keqiaozhang, 2018-08-02:
This issue cannot reproduced with latest code base, I tested it more than 20 times. so let's close this bug.

keqiaozhang, 2018-08-22:
This issue still can be reproduced during our test, the reproducibility is about 5%, reopen this bug to track this issue.

singalsu, 2018-09-04:
I don't have a 4ch output device. But I think I can add a level meter code with trace for each channel after volume and see if I can reproduce this with some nocodec topology.

tlauda, 2019-03-23:
@keqiaozhang Do we still have repro?

keqiaozhang, 2019-03-25:
@tlauda This issue is observed on a previous platform, this platform is not our POR any more and we don't have a device to support 4ch output now, so let's close this issue temporarily. we can reopen it if needed.

closed 2019-03-25 by keqiaozhang