thesofproject/sof #184: "DSP trace buffer overflow" in dmesg on APL using sof-apl-nocodec.tplg once boot up

zhuqingliang, 2018-08-02:
environment:
on APL
sof-dev: 08f7ddbcdd7eb4fcbf43cef09272215ef53dba7c
sof: c1123ec2ab994a3d1dadd2c64e4e352c75e88e0e
soft: cfff9e0b0402f0f7dd2bd9ba915fd6d1599b2fc8
tplg file name :
topology/spf-apl-nocodec.tplg

reproduces steps:
dmesg| grep error

log:
[dmesg.log](https://github.com/thesofproject/sof/files/2251767/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2251768/rmbox.log)

plbossart, 2018-08-02:
Liam and me saw a similar issue on GLK yesterday after adding only 4 data traces in the SSP code.

xiulipan, 2018-08-02:
@plbossart @lgirdwood 
This seems can not be avoid since we get too many trace log when setting up the ppl.
When the tplg first loaded in KERNEL, too many IPCs flushed into DSP and dma-buffer is overflow very fast.
Will check with @RanderWang after he is back about the up-limitation flow rate of the DMA-trace.

lgirdwood, 2018-08-02:
@xiulipan flow rate is fine, iirc problem is due to driver enabling DMA trace after topology has loaded.

plbossart, 2018-08-02:
@xiulipan we saw the message after adding a total of 8 traces, if it results in an overflow then something is very wrong with buffer management. We are not talking about massive bandwidth issues exceeding what the DMA hardware can do...

RanderWang, 2018-08-07:
Yes, I have made a solution to fix this issue. I will create PR after I refine it

zhuqingliang, 2018-08-08:
bug still exist in  31d8aa4c7a8a1656b9ead18d16a9625dae4ca1e1.

xiulipan, 2018-08-13:
fixed by #204