thesofproject/sof #192: DMIC:ipc error and DSP panic when doing pause / release

SinianYang, 2018-08-03:
This issue happens when run capture through dmic.

The reproducibility is almost 100%.
How to reproduce:
1. arecord -D hw:0,6 -c 2 -r 48000 -f 32_le  /dev/null -vv -i
2. pause / release 

Environment:
on UP^2
sof-dev: 08f7ddbcdd7eb4fcbf43cef09272215ef53dba7c
sof: c1123ec2ab994a3d1dadd2c64e4e352c75e88e0e
soft: cfff9e0b0402f0f7dd2bd9ba915fd6d1599b2fc8
tplg file name: sof-apl-nocodec.tplg
log:
[dmesg.log](https://github.com/thesofproject/sof/files/2255747/dmesg.log)
[memorywindow.log](https://github.com/thesofproject/sof/files/2255748/memorywindow.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2255749/rmbox.log)

singalsu, 2018-08-03:
You have an old rmbox binary since it doesn't show dmic trace texts.

singalsu, 2018-08-03:
I tried to repeat this but I don't get DSP panic. Does it happen with first space key press or do you need press pause a lot of times or very rapidly or something?

SinianYang, 2018-08-06:
@singalsu 
I just tested this issue and found out that when I used "-c 1" or "-c 2" in "arecord -D hw:0,6  -r 48000 -f 32_le /dev/null -vv -i" there was no such problem, but when I used "-c 4", once I first pressed space key, the dsp would panic. And I have already update reproduce steps and logs.

singalsu, 2018-08-10:
This trace output shows that the topology has configured DMIC for 2ch (num_pdm_active == 1, a pdm controller provides two channels so this should be 2 for 4ch) . It's no wonder that it fails. I think the bug is that this is not rejected cleanly.

```
0xf70 [0.085074]	delta [0.000002]	dmic dsc
0xf80 [0.085078]	delta [0.000004]	value 0x0000000000000001
0xf90 [0.085079]	delta [0.000001]	value 0x000000000007a120
0xfa0 [0.085081]	delta [0.000001]	value 0x0000000000493e00
0xfb0 [0.085082]	delta [0.000001]	value 0x000000000000bb80
0xfc0 [0.085083]	delta [0.000001]	value 0x0000000000000000
0xfd0 [0.085084]	delta [0.000001]	value 0x0000000000000020
0xfe0 [0.085086]	delta [0.000001]	value 0x0000000000000020
0xff0 [0.085087]	delta [0.000001]	value 0x0000000000000028
0x1000 [0.085089]	delta [0.000001]	value 0x000000000000003c
0x1010 [0.085090]	delta [0.000001]	value **0x0000000000000001**
```

keqiaozhang, 2018-08-17:
Since the DMIC is configured as 2ch in topology and this bug can only reproduce with 4ch, so let's close this bug.

closed 2018-08-17 by keqiaozhang