thesofproject/sof #198: rstrcmp - comparing different lengthed strings

mwierzbix, 2018-08-06:
In `src\lib\lib.c::rstrcmp(const char *s1, const char *s2)` no check is done for compared strings' lengths.

`rstrcmp("aa", "aab")` will return 0, whereas gcc implementation of `strcmp` will return -1.
Is this by design, or an oversight?

From C standard draft (https://port70.net/~nsz/c/c11/n1570.html#7.24.4.2) : 
> The sign of a nonzero value returned by the comparison functions `memcmp`, `strcmp`, and `strncmp `is determined by the sign of the difference between the values of the first pair of characters (both interpreted as unsigned char) that differ in the objects being compared.

Technically, null terminator is an inherent part of valid c-string.
So when comparing "aa" with "aab", when we compare following pairs
"a" with "a"
"a" with "a"
"\0" with "b" -> return -1.

Such additional check could look as follows:
```
/* did both string end */
if (*s1 != 0)
	return  1;
if (*s2 != 0)
	return -1;

/* match */
return 0;
```

plbossart, 2018-08-06:
Indeed looks like a bug to me. Good old strcmp issue...

mwierzbix, 2018-08-06:
C in a nutshell ;)
Will make a pull request in a moment.

mwierzbix, 2018-08-06:
Fix candidate @ https://github.com/thesofproject/sof/pull/199

closed 2018-08-07 by mwierzbix