thesofproject/sof #209: Random IPC + DSP status issues on startup (APL/LeafHill)

plbossart, 2018-08-07:
I sometimes get this sort of errors (DSP status in the weeds and unknown errrors). rebooting makes it go away, not 100% reproducible.

[    4.706216] sof-audio sof-audio: tplg: config SSP0 fmt 0x4001 mclk 24576000 bclk 1536000 fclk 48000 width (16)16 slots 2 mclk id 0
[    4.706234] sof-audio sof-audio: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.706250] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[    4.706267] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
[    5.013177] sof-audio sof-audio: error: ipc timed out for 0x80010000 size 0x50
[    5.013239] sof-audio sof-audio: status: fw entered - code 00000005
[    5.013300] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[    5.013352] sof-audio sof-audio: error: waking up any trace sleepers
[    5.013399] sof-audio sof-audio: error: failed to set DAI config for SSP0
[    5.013446] sof-audio sof-audio: ASoC: physical link loading failed
[    5.013489] sof-audio sof-audio: error: tplg component load failed -110
[    5.013546] sof-audio sof-audio: error: failed to load DSP topology -22
[    5.013592] sof-audio sof-audio: ASoC: failed to probe component -22
[    5.013653] sof-nocodec sof-nocodec: ASoC: failed to instantiate card -22
[    5.013738] sof-nocodec sof-nocodec: Topology forgot to remove link Port0?
[    5.013740] sof-nocodec sof-nocodec: Topology forgot to remove link Port1?
[    5.013742] sof-nocodec sof-nocodec: Topology forgot to remove link Port2?
[    5.013744] sof-nocodec sof-nocodec: Topology forgot to remove link Port3?
[    5.013745] sof-nocodec sof-nocodec: Topology forgot to remove link Port4?
[    5.013747] sof-nocodec sof-nocodec: Topology forgot to remove link Port5?
[    5.013749] sof-nocodec sof-nocodec: Topology forgot to remove link DMIC01?
[    5.013796] sof-nocodec: probe of sof-nocodec failed with error -22
[    5.107873] random: crng init done
[    5.107877] random: 7 urandom warning(s) missed due to ratelimiting
[    5.201053] sof-audio sof-audio: error: rx list empty but received 0x0
[    5.201119] sof-audio sof-audio: error: can't find message header 0x10000
[    5.695855] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
[    5.695873] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG

plbossart, 2018-08-08:
and another one after 200 load/unload without issues...

[   39.395133] sof-audio sof-audio: ipc tx: 0x30130000: GLB_TPLG_MSG: PIPE_COMPLETE
[   39.395160] sof-audio sof-audio: error : DSP panic!
[   39.395424] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[   39.395427] sof-audio sof-audio: status: fw entered - code 00000005
[   39.395791] sof-audio sof-audio: error: runtime exception
[   39.396078] sof-audio sof-audio: error: trace point 00004000
[   39.396382] sof-audio sof-audio: error: DSP Firmware Oops
[   39.396669] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore a
ccess
[   39.397305] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x000000a0 PS       0x00060025 SAR     0x00000001
[   39.397838] sof-audio sof-audio: EPC1     0xbe01dcda EPC2     0xbe011316 EPC3     0x00000000 EPC4    0x00000000
[   39.398364] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   39.398895] sof-audio sof-audio: EPS2     0x00060520 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[   39.399421] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[   39.399951] sof-audio sof-audio: stack dump from 0xbe07fe10
[   39.400247] sof-audio sof-audio: 0xbe07fe10: 0xbe01dcda 0x00000000 0x00000001 0x00000000
[   39.400671] sof-audio sof-audio: 0xbe07fe14: 0xbe050500 0x00000000 0xbe04e948 0x00000000
[   39.401102] sof-audio sof-audio: 0xbe07fe18: 0x0000000d 0x00000000 0xbe0245ea 0xbe024600
[   39.401528] sof-audio sof-audio: 0xbe07fe1c: 0x00cea498 0x00000000 0x00000020 0x00000000
[   39.401957] sof-audio sof-audio: 0xbe07fe20: 0x00000000 0x00000000 0xbe0478a0 0x00000000
[   39.402381] sof-audio sof-audio: 0xbe07fe24: 0x00000000 0x00000000 0x00000000 0x00000000
[   39.402811] sof-audio sof-audio: 0xbe07fe28: 0xc3700002 0x00000000 0x00110001 0x00000002
[   39.403235] sof-audio sof-audio: 0xbe07fe2c: 0x00cef853 0x00000000 0x00000002 0x00000000
[   39.403662] sof-audio sof-audio: error: waking up any trace sleepers
[   39.699041] sof-audio sof-audio: error: ipc timed out for 0x30130000 size 0xc
[   39.699431] sof-audio sof-audio: status: fw entered - code 00000005
[   39.699782] sof-audio sof-audio: error: runtime exception
[   39.700071] sof-audio sof-audio: error: trace point 00004000
[   39.700373] sof-audio sof-audio: error: DSP Firmware Oops
[   39.700664] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore a
ccess
[   39.701295] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x000000a0 PS       0x00060025 SAR     0x00000001
[   39.701824] sof-audio sof-audio: EPC1     0xbe01dcda EPC2     0xbe011316 EPC3     0x00000000 EPC4    0x00000000
[   39.702354] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   39.702880] sof-audio sof-audio: EPS2     0x00060520 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[   39.703405] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[   39.703932] sof-audio sof-audio: stack dump from 0xbe07fe10
[   39.704228] sof-audio sof-audio: 0xbe07fe10: 0xbe01dcda 0x00000000 0x00000001 0x00000000
[   39.704654] sof-audio sof-audio: 0xbe07fe14: 0xbe050500 0x00000000 0xbe04e948 0x00000000
[   39.705079] sof-audio sof-audio: 0xbe07fe18: 0x0000000d 0x00000000 0xbe0245ea 0xbe024600
[   39.705506] sof-audio sof-audio: 0xbe07fe1c: 0x00cea498 0x00000000 0x00000020 0x00000000
[   39.705931] sof-audio sof-audio: 0xbe07fe20: 0x00000000 0x00000000 0xbe0478a0 0x00000000
[   39.706357] sof-audio sof-audio: 0xbe07fe24: 0x00000000 0x00000000 0x00000000 0x00000000
[   39.706794] sof-audio sof-audio: 0xbe07fe28: 0xc3700002 0x00000000 0x00110001 0x00000002
[   39.707232] sof-audio sof-audio: 0xbe07fe2c: 0x00cef853 0x00000000 0x00000002 0x00000000
[   39.707677] sof-audio sof-audio: error: waking up any trace sleepers
[   39.708092] sof-audio sof-audio: tplg: complete pipeline PIPELINE.1.SSP0.OUT id 5

lgirdwood, 2018-08-08:
@plbossart this looks more like something on the firmware side that's causing the timout as the kernel log timestamps do show a timeout. Can you update the script to dump fw logs when timeout happens. Fwiw, this would be a good one for gdb when it's ready.

lgirdwood, 2018-08-08:
@plbossart btw, is this also seen on UP^2 ?

plbossart, 2018-08-09:
This happens on both LeafHill and Up2. Did you mean add a call to rmbox -t when the issue is detected?

lgirdwood, 2018-08-09:
@plbossart yes, lets dump the persistent error log

RanderWang, 2018-08-20:
The behavior of this issue is the same as https://github.com/thesofproject/sof/issues/180 with my commit.
So lets fix it together.