thesofproject/sof #210: Stress test : keeping pressing space key when playback and capture at the same time causes ipc timed out.

SinianYang, 2018-08-08:
This issue exists on apl and cnl.
Environment:
on up^2 using gcc
sof-v4.14: 14d7f16ce31d27f9171368ef0bfbbf00f6f160d4
sof: 31d8aa4c7a8a1656b9ead18d16a9625dae4ca1e1
soft: 91bb08dd8ce6305bda1649d0171c48456ba75421
tplg: sof-apl-nocodec.tplg

reproduce steps:
1. aplay -Dhw:0,0 -fS16_LE -c 2 -r 48000 /dev/zero  & arecord -Dhw:0,6 -f S32_LE -r 48000 -c 2 dmic.wav -vv -i
2. keep pressing space key

log:
[dmesg.log](https://github.com/thesofproject/sof/files/2269472/dmesg.log)
[memorywindow.log](https://github.com/thesofproject/sof/files/2269473/memorywindow.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2269474/rmbox.log)

Environment:
on up^2 using gcc
sof-dev: 82009079505a086987500df352b33bb600964dcc
sof: 31d8aa4c7a8a1656b9ead18d16a9625dae4ca1e1
soft: 91bb08dd8ce6305bda1649d0171c48456ba75421
tplg: sof-apl-nocodec.tplg

reproduce steps:
1. aplay -Dhw:0,0 -fS16_LE -c 2 -r 48000 /dev/zero  & arecord -Dhw:0,6 -f S32_LE -r 48000 -c 2 dmic.wav -vv -i
2. keep pressing space key until input/output error then ctrt+c
3. repeat step2 several times 

Environment:
on cannonlake using gcc
sof-dev: 82009079505a086987500df352b33bb600964dcc
sof: 31d8aa4c7a8a1656b9ead18d16a9625dae4ca1e1
soft: 91bb08dd8ce6305bda1649d0171c48456ba75421
tplg: test-ssp0-mclk-0-I2S-volume-s16le-s16le-48k-24000k-nocodec.tplg

reproduce steps:
1. aplay -Dhw:0,0 -fS16_LE -c 2 -r 48000 /dev/zero  & arecord -Dhw:0,0 -f S16_LE -r 48000 -c 2 test.wav -vv -i
2. keep pressing space key until input/output error then ctrt+c
3. repeat step2 several times 

log:
[dmesg.log](https://github.com/thesofproject/sof/files/2269475/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2269476/rmbox.log)

lgirdwood, 2018-08-08:
@RanderWang this is kernel bug related to IPC serialization.

plbossart, 2018-08-09:
Agree with @lgirdwood, there's definitively something fishy about IPCs (could be missing some interrupts since the code doesn't always use spin_lock_irqsave

xiulipan, 2018-08-15:
@lgirdwood @plbossart 
I found that we may have some error inside the firmware.
When I reproduced one IPC timeout and dig into the trace, I found that the DSP come to alive after few mintes.

Here is the demsg logged the back to live DSP.
The DSP got into hang in 60s, and suddenly got back to live at 204s with a interrupt from 
CAPTURE stream from function
`irqreturn_t hda_dsp_stream_threaded_handler(int irq, void *context)`

It seems we have some error with our capture stream.

> [   59.892889] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
> [   59.892909] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
> [   59.912415] sof-audio sof-audio: pcm: trigger stream 0 dir 0 cmd 0
> [   59.912460] sof-audio sof-audio: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
> [   60.216286] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
> [   60.216306] sof-audio sof-audio: status: fw entered - code 00000005
> [   60.216332] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
> [   60.216339] sof-audio sof-audio: error: waking up any trace sleepers
> [   60.216355]  Passthrough: ASoC: trigger FE failed -110
> [   60.216393] sof-audio sof-audio: pcm: free stream 0 dir 0
> [   60.216427] sof-audio sof-audio: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
> [   60.520283] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
> [   60.520303] sof-audio sof-audio: status: fw entered - code 00000005
> [   60.520329] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
> [   60.520336] sof-audio sof-audio: error: waking up any trace sleepers
> [   60.520489] sof-audio sof-audio: pcm: close stream 0 dir 0
> **[  204.888536] sof-audio sof-audio: cstream 0 status 0x24**
> [  204.888588] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
> [  204.888608] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
> [  205.383422] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
> [  205.383441] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
> [  205.878211] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG

lgirdwood, 2018-08-15:
@xiulipan so you mean that FW blocks/spins/sleeps on something for 144s and then returns to normal ?
The capture IRQ is probably freeing a resource/lock ?

Might be worth adding some debug to IPC IRQ handler in FW to trace each message received. That way you can check if IPC IRQ is received for timeout message and work from there. Remeber IPC message is recived at IRQ level 2 but processed later on at level 0 (so can be pre-emptied).

xiulipan, 2018-08-16:
@lgirdwood 
I am not sure about that. This bug is totally random and hard to reproduce if you add some log.
I will try to analysis the pattern like you said.

I still could not tell if the final TRIGGER_STOP ipc cause the DSP hang or the DSP hang make the ALSA send a STOP to DSP. I think it should be the DSP in hang first and no IPC message could be received then. I will have some log to figure this first as you recommend.

xiulipan, 2018-08-17:
After apply #232 and https://github.com/thesofproject/linux/pull/88
The reproduce bug will not be influence by the #180.

After enable TRAVEV and TRACEM, on APL, bootup will failed with some tplg ipc timeout. But the DSP is not dead since DMA TRACE is continue on later.

> [    3.291083] sof-audio sof-audio: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
> [    3.465697] Adding 4038652k swap on /dev/mmcblk0p3.  Priority:-2 extents:1 across:4038652k SS
> [    3.596631] sof-audio sof-audio: error: ipc timed out for 0x30030000 size 0x10
> [    3.596681] sof-audio sof-audio: status: fw entered - code 00000005
> [    3.596732] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
> [    3.596772] sof-audio sof-audio: error: waking up any trace sleepers
> [    3.596811] sof-audio sof-audio: error: failed to add route sink SSP5.OUT control none source BUF11.1
> [    3.596880] sof-audio sof-audio: sink BUF12.1 control none source SSP5.IN
> [    3.785947] sof-audio sof-audio: error: rx list empty but received 0x0
> [    3.785955] sof-audio sof-audio: error: can't find message header 0x30030000
> [    3.785981] sof-audio sof-audio: ipc rx: 0x90020000: GLB_TRACE_MSG
> [    3.785996] sof-audio sof-audio: ipc rx done: 0x90020000: GLB_TRACE_MSG
> [    3.904041] sof-audio sof-audio: error: ipc timed out for 0x30030000 size 0x10
> [    3.904052] sof-audio sof-audio: status: fw entered - code 00000005
> [    3.904072] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
> [    3.904074] sof-audio sof-audio: error: waking up any trace sleepers
> [    3.904082] sof-audio sof-audio: error: failed to add route sink BUF12.1 control none source SSP5.IN
> [    3.904111] sof-audio sof-audio: sink BUF13.0 control none source DMIC0.IN
> [    3.904157] sof-audio sof-audio: ipc tx: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
> [    3.905132] sof-audio sof-audio: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
> [    3.905172] sof-audio sof-audio: tplg: config SSP0 fmt 0x4001 mclk 24576000 bclk 1536000 fclk 48000 width (16)16 slots 2 mclk id 0
> [    3.905295] sof-audio sof-audio: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
> [    3.905403] sof-audio sof-audio: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG

This should be the same issue faced with @RanderWang in #180

RanderWang, 2018-08-24:
I also debug this issue. Now after thousand cycles of pause-resume, the DSP is no response to host and IPC timeout. I found that the DSP hang when resumed from paused status.

lgirdwood, 2018-08-24:
@RanderWang resuming during pause state is a different and new issue. Could be that the driver has not restored context correctly.

IPC race with BUSY bit has been identified on APL and CNL and @xiulipan is doing PR.

RanderWang, 2018-08-28:
call stack after ipc timeout:

[   52.800377] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
[   52.800928] sof-audio sof-audio: status: fw entered - code 00000005
[   52.801399] sof-audio sof-audio: error: unexpected fault 0x00000007 trace 0x00004000
[   52.801952] sof-audio sof-audio: error: unknown reason 00000007
[   52.802377] sof-audio sof-audio: error: trace point 00004000
[   52.802784] sof-audio sof-audio: error: DSP Firmware Oops
[   52.803173] sof-audio sof-audio: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[   52.803832] sof-audio sof-audio: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00060725 SAR     0x0000001b
[   52.804550] sof-audio sof-audio: EPC1     0xbe0124e7 EPC2     0xbe010c2b EPC3     0xbe01394b EPC4    0x00000000
[   52.805268] sof-audio sof-audio: EPC5     0xbe011bb3 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   52.805986] sof-audio sof-audio: EPS2     0x00060520 EPS3     0x00060b22 EPS4     0x00000000 EPS5    0x00060922
[   52.806703] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000001d8 INTERRU 0x00000232
[   52.807436] sof-audio sof-audio: stack dump from 0xbe07fe70
[   52.807839] sof-audio sof-audio: 0xbe07fe70: 0x0f50af35 0x00000000 0xbe051100 0x00000000
[   52.808418] sof-audio sof-audio: 0xbe07fe74: 0x7e011bb3 0x00000000 0xffffffff 0x00000000
[   52.808997] sof-audio sof-audio: 0xbe07fe78: 0x00001634 0x00000000 0xbe051100 0xbe050f00
[   52.809575] sof-audio sof-audio: 0xbe07fe7c: 0xbe01cd10 0x00000000 0xbe0468c4 0x00000000
[   52.810153] sof-audio sof-audio: 0xbe07fe80: 0xbe0108ae 0x00000000 0x00000000 0x00000000
[   52.810732] sof-audio sof-audio: 0xbe07fe84: 0xbe04eb48 0x00000000 0x00000010 0x00000000
[   52.811310] sof-audio sof-audio: 0xbe07fe88: 0xbe0468c0 0x00000000 0x000000e0 0x00000000
[   52.811889] sof-audio sof-audio: 0xbe07fe8c: 0x0f50af6b 0x00000000 0xbe022629 0x00000000

RanderWang, 2018-08-28:
EPC1 is in <memcpy>
EPC2 is in  <do_task>
EPC3 is in  <platform_pm_runtime_put>
EPC5 is in <schedule_task_complete>

RanderWang, 2018-08-28:
I add trace log in dai_dma_cb or in dw_dma_irq_handler, no ipc timeout.
I also try to  enable tracev to get some hint. But I didn't get ipc timeout with tracev enabled.
So it is difficult for me to get hint by trace log.

RanderWang, 2018-08-28:
I also call arch_interrupt_get_enabled to dump interrupt enabled register at failed time and at normal time. Both value are 0x1d8.

RanderWang, 2018-08-29:
@lgirdwood  they are included in the kernel panic dump

RanderWang, 2018-08-29:
@tlauda  yes, good idea, I will try

RanderWang, 2018-08-29:
@lgirdwood  @tlauda  @xiulipan 

Hi all,  I work with xiuli find there is a deadloop in schedule function and it is caused by a task is enqueued two times. After fixing this issue, the bug is killed

lgirdwood, 2018-08-29:
@RanderWang great work :)

xiulipan, 2018-08-30:
@lgirdwood 

> Are you saying the list loops forever under certain circumstances ?

yes, the list is broken since we try to append one list item twice into the list. But the list item shares the same **next** and **prev** pointer, which make the list to be a chaos. This is what you have mentioned in #272 
> I've spotted a possible race we need to resolve in scheduler. We need to add a check to make sure we are not appending a task to the list twice.

For the list item del
> Also look at where we add the task to the task list. We keep appending the list item and never remove it. We need to check whether then task is on the list before we add it.

Actually we will remove the task as the task is complete.
```c
/* Remove a task from the scheduler when complete */
void schedule_task_complete(struct task *task)
{
	uint32_t flags;

	tracev_pipe("com");

	spin_lock_irq(&sch->lock, flags);
	list_item_del(&task->list);
	task->state = TASK_STATE_COMPLETED;
	spin_unlock_irq(&sch->lock, flags);
}
```
And this is called in 
```c
static void _irq_task(void *arg)
{
	struct irq_task *irq_task = *(struct irq_task **)arg;
	struct list_item *tlist;
	struct list_item *clist;
	struct task *task;
	uint32_t flags;

	/* intentionally don't lock list to have task added from schedule irq */
	list_for_item(tlist, &irq_task->list) {
		task = container_of(tlist, struct task, irq_list);

		if (task->func)
			task->func(task->data);

		schedule_task_complete(task);
	}
...
}
```
from `arch_run_task`

RanderWang, 2018-08-31:
@lgirdwood @tlauda  @xiulipan 

Hi all,  I don't think the deadloop is the root reason of this bug. It is a issue but never happen in normal case. It is caused by my change. I scheduled the pipe after ipc timeout.

I add many logs and confirm it is still caused by dai interrupt missing. I tuned my test case and made the IPC timeout at a fix point. The following is the fixed log.

0x0 [3.111575]	delta [3.111575]	ipc Pen
0x10 [5.678263]	delta [2.566688]	pipe run
0x20 [5.678267]	delta [0.000003]	pipe ept
0x30 [5.678268]	delta [0.000001]	pipe pxr // playback XRUN recovery
0x40 [5.678288]	delta [0.000020]	dma DDi
0x50 [5.678294]	delta [0.000005]	dai xrc   //capture interrupt
0x60 [5.678299]	delta [0.000006]	pipe run
0x70 [5.678429]	delta [0.000130]	pipe fxr // playback XRUN recovery successfully
0x80 [5.678431]	delta [0.000002]	pipe stc
0x90 [5.678433]	delta [0.000001]	pipe ept
0xa0 [5.678474]	delta [0.000041]	pipe epe
0xb0 [5.678476]	delta [0.000002]	pipe stc
0xc0 [5.678483]	delta [0.000007]	pipe twk
0xd0 [5.679289]	delta [0.000806]	dai xrc  //capture interrupt
0xe0 [5.679295]	delta [0.000006]	pipe run
0xf0 [5.679299]	delta [0.000004]	pipe ept
0x100 [5.679342]	delta [0.000043]	dai xrp    //playback interrupt
0x110 [5.679347]	delta [0.000005]	pipe run
0x120 [5.679352]	delta [0.000005]	pipe epe
0x130 [5.679353]	delta [0.000002]	pipe stc
0x140 [5.679355]	delta [0.000001]	pipe ept
0x150 [5.679396]	delta [0.000041]	pipe epe
0x160 [5.679398]	delta [0.000002]	pipe stc
0x170 [5.679399]	delta [0.000002]	pipe tsk  // _irq_task complete
                                                             // no more interrupt

And if I add code at the end of the in _irq_task to check the DMA engine
                for(...)
			trace_error_value(dw_read(dma, DW_SAR(0)));
			trace_error_value(dw_read(dma+1, DW_DAR(1)));

more DMA irq would come.

0x170 [84.430771]	delta [0.000002]	pipe stc
0x180 [84.430773]	delta [0.000002]	ipc tsk
0x190 [84.430894]	delta [0.000121]	value 0x00000000be001324
0x1a0 [84.430896]	delta [0.000001]	value 0x00000000be00206c
0x1b0 [84.431017]	delta [0.000121]	value 0x00000000be00133c
0x1c0 [84.431018]	delta [0.000001]	value 0x00000000be002084
0x1d0 [84.431140]	delta [0.000121]	value 0x00000000be001350
0x1e0 [84.431141]	delta [0.000001]	value 0x00000000be00209c
0x1f0 [84.431262]	delta [0.000121]	value 0x00000000be001368
0x200 [84.431264]	delta [0.000001]	value 0x00000000be0020b0
0x210 [84.431385]	delta [0.000121]	value 0x00000000be00137c
0x220 [84.431386]	delta [0.000001]	value 0x00000000be0020c8
0x230 [84.431508]	delta [0.000121]	value 0x00000000be00137c
0x240 [84.431509]	delta [0.000001]	value 0x00000000be0020e0
0x250 [84.431630]	delta [0.000121]	value 0x00000000be00137c
0x260 [84.431632]	delta [0.000001]	value 0x00000000be0020f8
0x270 [84.431662]	delta [0.000030]	dai xrc
0x280 [84.431667]	delta [0.000005]	pipe run
0x290 [84.431706]	delta [0.000039]	dai xrp
0x2a0 [84.431712]	delta [0.000005]	pipe run
0x2b0 [84.431774]	delta [0.000062]	value 0x00000000be00124c
0x2c0 [84.431775]	delta [0.000001]	value 0x00000000be001f94
0x2d0 [84.431897]	delta [0.000122]	value 0x00000000be001264
0x2e0 [84.431898]	delta [0.000001]	value 0x00000000be001fac
0x2f0 [84.432019]	delta [0.000121]	value 0x00000000be00127c
0x300 [84.432021]	delta [0.000001]	value 0x00000000be001fc4
0x310 [84.432022]	delta [0.000002]	pipe ept
0x320 [84.432064]	delta [0.000042]	pipe epe
0x330 [84.432066]	delta [0.000002]	pipe stc
0x340 [84.432067]	delta [0.000001]	pipe ept
0x350 [84.432108]	delta [0.000041]	pipe epe
0x360 [84.432110]	delta [0.000002]	pipe stc

And FW again stop after pipe scheduled.

So it seems that DMA interrupt is blocked by something. I confirmed that the interrupt status is ok. 
And I can trigger ssp interrupt and schedule interrupt after FW stop.

Actually, it is very difficult to reproduce this issue. Sometimes you need to wait a long time to get the bug. And when you add some logs to get FW info , the issue cant be reproduced.

lgirdwood, 2018-08-31:
@RanderWang Scheduler, workQ and DMA are all level 5 interrupts, meaning they cannot pre-empt each other. e.g if scheduler is running then DMA IRQ will not get services until scheduler IRQ completes. Please align with @xiulipan and mkae sure you also have latest scheduler & task testing updates.

RanderWang, 2018-08-31:
@lgirdwood  but IRQ_task run on lower level than dma interrupt ?  I also check the SSP setting when IPC timeout. all the ssp registers are correct. And I check the output of I2S pin, both bit clock and sync clock are
correct. And ssp is also a level 5 interrupt. If I set sss register to test ssp interrupt when ipc timeout, dma interrupt is triggered every 2us. So I don't know why so many dma interrupts

lgirdwood, 2018-08-31:
@RanderWang  irq_task can run at multiple levels and is platform dependent. If the IRQ task priority < scheduler/workQ then it can be preempted. If you are seeing DMA irqs every 2us then the DMA irq is not being cleared. Please update with @xiulipan as there is a race that causes list corruption, leading to potential execution of random code (and hence you may never clear DMA IRQ).

RanderWang, 2018-09-03:
@lgirdwood  @tlauda 
fix by https://github.com/thesofproject/sof/pull/292

ZhendanYang, 2018-09-07:
This issue can't be reproduced on APL and CNL with stable-1.2 branch(commit 251e68f291853e3c22ce617f23f94c01231268db).

xiulipan, 2018-09-13:
APL fixed with #367 #368 on master branch

ZhendanYang, 2018-09-14:
Latest status:
Test with stable 1.2 branch, APL and CNL have no such issue.
Test with master branch, on APL, there has DSP panic:

[  707.987916] sof-audio sof-audio: error : DSP panic!
[  707.987924] sof-audio sof-audio: panic: dsp_oops_offset 788480 offset 788480
[  707.987928] sof-audio sof-audio: status: fw entered - code 00000005
[  707.987949] sof-audio sof-audio: error: runtime exception
[  707.987952] sof-audio sof-audio: error: trace point 00004000
[  707.987955] sof-audio sof-audio: error: DSP Firmware Oops
[  707.987958] sof-audio sof-audio: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  707.987961] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x00000028 PS       0x00070025 SAR     0x00000019
[  707.987963] sof-audio sof-audio: EPC1     0xbe01cb86 EPC2     0xbe010d11 EPC3     0xbe0122bd EPC4    0x00000000
[  707.987966] sof-audio sof-audio: EPC5     0xbe010d11 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  707.987969] sof-audio sof-audio: EPS2     0x00060720 EPS3     0x00060022 EPS4     0x00000000 EPS5    0x00060720
[  707.987971] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x000101d8 INTERRU 0x00000222
[  707.987974] sof-audio sof-audio: stack dump from 0xbe07fd50

[dmesg.log](https://github.com/thesofproject/sof/files/2381961/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2381962/rmbox.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2381963/rmbox-t.log)

Test with stable1.2, on CNL, there has input/output error, no ipc timeout, no DSP panic, and can recover itself.

Test env:
kernel: sof-dev ae5f868
sof: matster f17abe6
soft: d77beaf

PR #373 (2018-09-17): Merge stable 1.2 task and scheduler fix into master
test and rebase the following patch from stable-1.2 to master to fix #210 
1. fix some schedule bugs
2. add priority support into scheduler and task
3. refine arch task to avoid double run issues

ipc: set IPC process task to lower priority
scheduler: add task priority into scheduler
schedule: fix schedule list double add
schedule: fix edf reschedule
pipe: fix a infinite loop issue in schedule module
task: refine task to avoid double task function run
closed 2018-09-17 by lgirdwood