thesofproject/sof #215: tdf8532 codec arecord noise

SinianYang, 2018-08-09:
Summary: Audios recorded by arecord are full of noise but no recorded  content.  This issue doesn't happen every time but alternately.

reproduce steps:
1. arecord -D hw:0,1 -c 8 -r 48000 -f s32_le test.wav -vv -i
2. play test.wav in audacity.

Environment:
on apl using gcc
sof-dev: 82009079505a086987500df352b33bb600964dcc
sof: f4716b12b9bfd5eb5e5e15ca6a5547467df88046
soft: ad895a8d00efc393db24d2b37322e6a07296c356
tplg: sof-apl-tdf8532.tplg

log: 
[dmesg.log](https://github.com/thesofproject/sof/files/2273769/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2273770/rmbox.log)

lgirdwood, 2018-08-09:
@SinianYang this is a driver/codec bug, not firmware. Please move to kernel.

plbossart, 2018-08-09:
@SinianYang did you mean the error occur exactly 50% of the time, in a pass/fail/pass/fail pattern?

zhigang-wu, 2018-08-10:
the TDF8532 Codec did not support the capture feature. 
why you do this test?
Could you confirm the device you used? it is enumerated for the DIRANA?
If yes, please correct the title of this bug report.

zhigang-wu, 2018-08-10:
@lgirdwood 
At present, we cannot confirm this issue caused by which part.
But It must be a regression issue.

zhigang-wu, 2018-08-10:
the capture stream path is:
MIC -> DIRANA -> SSP2(DSP) -> Driver(Kernel)
maybe it is caused by firmware.

SinianYang, 2018-08-10:
@plbossart  sometimes in a fail/pass/fail/pass pattern.
I update the issue summary and upload a test wav sample.

SinianYang, 2018-08-10:
This error happens every time now at latest code base :
sof-dev: 2de67aacaa795ed216b176dbee799f0ec18d6403
sof: 597194f3dd8f7899384d0819548e05f75257ad15
soft: fe1f3ba18072ab70697d00648cd110a3131c12b4
tplg: sof-apl-tdf8532.tplg

keqiaozhang, 2018-08-21:
I modified the topology to support 2ch or 4ch on SSP2, both of them can get clear sound without such noise, this issue only happens with 8ch capture.

zhigang-wu, 2018-08-21:
@keqiaozhang 
you can do a test to confirm one point: whether the 8-channel volume component has problem.
Do the loopback on the UP^2 on tdm8 mode. to make sure the captured result is OK or not.

mengdonglin, 2018-09-10:
Close this issue for no customer attention and will re-open if situation changes.
We will cover 8-channel capture in other platforms for the moment.

closed 2018-09-10 by mengdonglin