thesofproject/sof #222: Add support for GCC 7/ -Wimplicit-fallthrough=3

plbossart, 2018-08-10:
Linux kernel folks have started detecting implicit fallthroughs in switch cases, and it's becoming a requirement to explicitly add a comment to state the fallthrough is intentional (no missing break statement).
This is supported by GCC7, SOF should make use of this more recent compiler to improve the code.

xiulipan, 2018-08-13:
reference here
https://developers.redhat.com/blog/2017/03/10/wimplicit-fallthrough-in-gcc-7/
@plbossart 
Should we try to build Linux with GCC 4.8 to GCC 8.1 to avoid potential issues?

singalsu, 2018-09-04:
I have gcc version 7.3.0 so I can start with this.

singalsu, 2018-09-07:
@lgirdwood I will introduce a PR that enforces -Wimplicit-fallthrough=3 for SOF rimage and host builds. I assume there's no GCC7 for xtensa available so I'll not change it yet. There was one fix need for rimage (add comment that's sufficient for level 3 check.)

singalsu, 2018-09-07:
This is addressed by PR #323

tlauda, 2018-10-23:
@lgirdwood @singalsu Should be closed?

singalsu, 2018-10-24:
@tlauda @lgirdwood Yep, this check has been enabled already for a while.

closed 2018-10-24 by singalsu