thesofproject/sof #225: Trace error: "dma eI2" when doing pause/release

keqiaozhang, 2018-08-14:
This issue happens on BYT when playback or capture. but it doesn't affect audio.  The reproducibility is low < 5%.
[rmbox-t.log](https://github.com/thesofproject/sof/files/2287023/rmbox-t.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2287025/rmbox.log)

How to reproduce:
1. aplay -Dhw:0,0 -f dat -c 2 /dev/zero & arecord -D hw:0,0 -f dat -c 2 /dev/null -i 
2. press the space key to do pause/release for capture.
3. check the rmbox

Error log:
using 19.20MHz timestamp clock
0x0 [15512.024334]      delta [********]        dma eI2
0x10 [15512.024337]     delta [0.000002]        value 0x0000000000000001
0x20 [2783.935138]      delta [********]        dma eI2
0x30 [2783.935140]      delta [0.000002]        value 0x0000000000000002
0x40 [2783.936138]      delta [0.000998]        dma eI2
0x50 [2783.936140]      delta [0.000002]        value 0x0000000000000002
0x60 [2783.937138]      delta [0.000998]        dma eI2
0x70 [2783.937140]      delta [0.000002]        value 0x0000000000000002

lgirdwood, 2018-08-14:
@xiulipan lets also dump the error code if we can. It's probably harmless notification and could be verbose trace level.

xiulipan, 2018-08-15:
@lgirdwood 
will done. I believe this is because BYT will have three channel enable with only one IRQ handler.
Let me try to enable all SSP on BYT and have a test.

mengdonglin, 2018-10-10:
@ZhendanYang could you check if this issue is still there with new logger?

ZhendanYang, 2018-10-10:
@mengdonglin BYT is in bad health at the moment. I can't check this issue for now. I'll check this issue as soon as BYT become healthy.

lgirdwood, 2018-10-16:
@ZhendanYang I cant find "el2" in the source. Can you re-test with master.

ZhendanYang, 2018-10-17:
Blocked by #443 . And rmbox and logger are both empty.

tlauda, 2018-11-28:
@ZhendanYang Any update?

xiulipan, 2018-11-28:
@tlauda 
I think we can close this issue. Since no more rmbox is working now.

closed 2018-11-28 by xiulipan