thesofproject/sof #228: Replace autotools with Kconfig

lgirdwood, 2018-08-15:
Replace GNU autotools with Linux Kconfig for SOF build and configuration.

jajanusz, 2018-08-24:
I can do that, but I have already other tasks assigned, so idk if I'll be able to do it before 1.3 release date (Oct 1st).

lgirdwood, 2018-08-24:
@jajanusz great, we should also create a config for each platform too.

lgirdwood, 2018-08-30:
@jajanusz @ mmaka1 pre-packed that we could use. Can also be used on Windows.

http://ymorin.is-a-geek.org/projects/kconfig-frontends

jajanusz, 2018-12-04:
@lgirdwood 
For initial version I want to support targets that we already use:
```
make
make rimage
make bin
make install
make doc
make check/test
```

+What's below ->

Targets related to Kconfig:
```
Cleaning targets:
  clean           - Remove most generated files but keep the config
  mrproper        - Remove all generated files + config + various backup files

Configuration targets:
  config          - Update current config utilising a line-oriented program
  nconfig         - Update current config utilising a ncurses menu based program
  menuconfig      - Update current config utilising a menu based program
  xconfig         - Update current config utilising a Qt based front-end
  gconfig         - Update current config utilising a GTK+ based front-end
  oldconfig       - Update current config utilising a provided .config as base
  defconfig       - New config with default from ARCH supplied defconfig
  savedefconfig   - Save current config as ./defconfig (minimal config)
  allnoconfig     - New config where all options are answered with no
  allyesconfig    - New config where all options are accepted with yes
  allmodconfig    - New config selecting modules when possible
  alldefconfig    - New config with all symbols set to default
  randconfig      - New config with random answer to all options
  listnewconfig   - List new options
  olddefconfig    - Same as oldconfig but sets new symbols to their
                    default value without prompting
```
Kernel-like make parameters:
```
make V=0|1 [targets] 0 => quiet build (default), 1 => verbose build
make V=2   [targets] 2 => give reason for rebuild of target
make O=dir [targets] Locate all output files in "dir", including .config
# Other paramters like C=, W= may be added later
```
Arch-specific targets, like:
```
apollolake_defconfig - Build for APL
cannonlake_defconfig - Build for CNL
#...
#It's similar to what's you get for make ARCH=x86 help in kernel, f.e.:
  i386_defconfig           - Build for i386
  x86_64_defconfig         - Build for x86_64
```

If you need something else to be done kernel-like way or some that's specific to our project please let me know.

lgirdwood, 2018-12-05:
Looks good to me :) We probably also need "make tools" instead or make rimage as this would include all tools and also "make topology" (@bkokoszx is splitting topology out of tools to make this easier).

jajanusz, 2018-12-05:
I'd leave tools as they are for initial support, they have their own buildsystem anyway and we can quickly port them to kconfig after it's done for FW. Rimage is a bit different, tbh it should be just built as hostprogs for `make bin` target, `rimage` target is going to be somewhere in makefile but there should be no need to invoke it explicitly.

jajanusz, 2019-01-24:
Added in #845

closed 2019-01-24 by jajanusz

lgirdwood, 2019-01-25:
Good job @jajanusz !