thesofproject/sof #230: IPC error after booting on cnl

SinianYang, 2018-08-17:
Summary: There are ipc error in dmesg after booting on cnl.

Environment:
sof-dev: 5420b87addf10a074fb81a555dff05a4a177af93
sof: bfc7e467f7b283d0b7db5dff7321c8c5ec400980
soft: a01d56f852d0b0fa865b666e5e7acc4baee1e1be
tplg file: sof-cnl-rt274,tplg

Logs:
[dmesg.log](https://github.com/thesofproject/sof/files/2296513/dmesg.log)
[rmbox.log](https://github.com/thesofproject/sof/files/2296514/rmbox.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2296515/rmbox-t.log)

lgirdwood, 2018-08-17:
@SinianYang this looks like Pulseaudio trying lots of different hw_params() and is expected. Not an error unless all hw_params are rejected, in that case it's a topology bug.

xiulipan, 2018-08-17:
@lgirdwood 
If this is some issues with our `SectionPCMCapabilities` in tplg.
Should we not pass some un-supported in tplg for diver?

SinianYang, 2018-08-17:
@lgirdwood  If I turn off Pulseaudio, there is no ipc error. Once I run pulseaudio --start, ipc error occurs. And paplay, parecord can't work normally.

mengdonglin, 2018-08-31:
We only have 1 CNL RVP used by validation team. Drop this when we have new GLV board?

mengdonglin, 2018-08-31:
Close this bug since we no longer support ALC274 codec but use no codec mode to test CNL platform.

closed 2018-08-31 by mengdonglin