thesofproject/sof #231: ALSA XRUN happen when playback or capture on each platform

SinianYang, 2018-08-17:
ALSA XRUN will random happen when playback or capture on APL, BYT, CNL.
This issue will be reproduce with high possibility with BUFFER_SIZE/PERIOD_SIZE=2
But audio functions are not affected by this issue.
This issue need developer to comment.
@xiulipan

lgirdwood, 2018-08-17:
@SinianYang we are missing a whole lot of details here. Logs, commit IDs, command line used in test etc.

lgirdwood, 2018-08-17:
@xiulipan I need to rule out if this is FS related as eMMC has high latency. i.e. does playing /dev/zero work ? or capturing to /dev/null ?

xiulipan, 2018-08-17:
@lgirdwood 
This issue is here to track later findings.
I think we did not do this test now. Will update the result here later.

SinianYang, 2018-08-21:
@lgirdwood 
There is no xrun when playing /dev/zero or capturing to /dev/null. But if I play /dev/zero and capture /dev/null at the same time, both overrun and underrun occur.

mengdonglin, 2018-09-17:
Remove branch label br-glk because all generic fixes on glk-003 branch have been merged by master.  So testing with master branch will be okay for mature platforms. glk release branch does not focus on compatibility across platforms.

mengdonglin, 2018-09-30:
@ZhendanYang @bardliao Let's check this issue after APL RVP is set up.