thesofproject/sof #238: IPC IRQ may be handled outside wait_for_interrupt

xiulipan, 2018-08-21:
In the src/tasks/audio.c

```c
	while (1) {

		/* sleep until next IPC or DMA */
		sa_enter_idle(sof);
		wait_for_interrupt(0);

		/* now process any IPC messages from host */
		ipc_process_msg_queue();

		/* schedule any idle tasks */
		schedule();
	}
```

`wait_for_interrupt` expect an interrupt and then `ipc_process_msg_queue` can process the IPC.
But if IPC interrupt is handled in other place, eg(in `schedule`), and we may have to wait for another interrupt to make the `ipc_process_msg_queue` go. This will cause the IPC timeout from HOST dmesg.

This should be the root cause of #209
And new issue of #180

xiulipan, 2018-08-21:
@lgirdwood 
Test with some flags, it will make the DSP busy with handling the endless IRQ. The flags will not disable the IRQ, DSP could not control when the IRQ comes. A flag will not fix this issue
@RanderWang 
We may need to mask the interrupt rather then clear or check it. This may solve the problem caused by register clean limitation.

lgirdwood, 2018-08-21:
@xiulipan yes IRQ will need masked. You could alter like wait_for_interrupt(level, irq_unmask), that way wfi would unmask the IRQ prior to entering WFI.

xiulipan, 2018-08-24:
@lgirdwood 
Let me explain the similar situation of IPC/IPC process and DMA/pipeline schedule copy.

```c
  	while (1) {

		/* sleep until next IPC or DMA */
		sa_enter_idle(sof);
		wait_for_interrupt(0); // DMA IRQ and IPC IRQ are expected to be handled here

		/* now process any IPC messages from host */
		ipc_process_msg_queue(); // IPC process

		/* schedule any idle tasks */
		schedule(); // schedule work to run
	}
```

If one DMA IRQ is handled outside WFI, then WFI won't exit and can not go into function `schedule()`.
So the scheduled pipeline copy task can not be executed until next IRQ comes. As the pipeline copy has more strict timeline request there may be some xrun for the dai.

lgirdwood, 2018-08-24:
@xiulipan DMA DAI IRQ calls schedule() directly to schedule all pipeline work, so it's guranteed to always run. The difference with IPC is that we currently delay the processing to this loop (by using a flag), but that can race. The solution is to schedule_idle() the IPC work as a task that should be executed when next idle.

xiulipan, 2018-08-24:
@lgirdwood 
Sorry about the mistake, I find I misread the code.
Yes the DMA IRQ is not like the IPC IRQ.

> schedule_idle() the IPC work as a task that should be executed when next idle.
But waht if the WFI did not got another IRQ intime, and wait to the IPC timeout?

I think we can jump this waiti if we have unhandled IPC message.

xiulipan, 2018-08-27:
@lgirdwood 
I have a patch that enable the schedule_idle solution. But it seems there still be the similar error.

>   	while (1) {
> 
> 		/* sleep until next IPC or DMA */
> 		sa_enter_idle(sof);
> 		wait_for_interrupt(0); // IPC IRQ are expected to be handled here
> 
> 		/* now process any IPC messages from host */
> 		/* ipc_process_msg_queue(); // IPC process now in schedule */
> 
> 		/* schedule any idle tasks */
> 		schedule(); // schedule idle work to run
> 	}

If some IPC IRQ happens before WFI and schedule a task, then WFI will wait for some IRQ to exit the WFI. Then `schedule()` will schedule the idle task, but the IPC process will timeout at that time.

I think we may need some timer work like the DMA_TRACE to set some timer IRQ for the IPC handler

lgirdwood, 2018-08-27:
@xiulipan give the task a deadline when you schedule it. It will be scheduled before this deadline.

xiulipan, 2018-08-28:
@lgirdwood 
I tried to use `schedule_task_idle` with timeout of 100ms. But the test result seems to be no effect.
I found that the deadline won't take effects until the `schedule` function is called.

The problem now is that we can't not exit WFI if some IRQ is handled outside WFI. The task uses software IRQ that could not solve our problem. I am trying to use `work` with timer to have some scheduled work to handle IPC. Work will trigger some timer IRQ to exit the WFI just like the DMA-TRACE done.

PR #268 (2018-08-28): ipc: use scheduler task for IPC process
IPC process is handled after wait_for_interrupt, but there may some IPC
is handled outside this WFI function. Use scheduler to avoid wait for
IRQ caused IPC timeout.

fix #238
closed 2018-08-28 by lgirdwood