thesofproject/sof #257: rimage: strcpy possible buffer overflow

plbossart, 2018-08-23:
This code in pkcs1_5.c is flagged by Coverity as problematic
```c
	if (!image->key_name)
		sprintf(path, "%s/otc_private_key.pem", PEM_KEY_PREFIX);
	else
		strcpy(path, image->key_name);
```
CID 313456 (#1 of 1): Copy into fixed size buffer (STRING_OVERFLOW)
3. fixed_size_dest: You might overrun the 256-character fixed-size string path by copying image->key_name without checking the length.
   	4. parameter_as_source: Note: This defect has an elevated risk because the source argument is a parameter of the current function

xiulipan, 2018-08-24:
@plbossart i will thake #257 and #258.
If Covertiy is some new scan? and how could I re-run to check if this could solve the problem.

plbossart, 2018-11-02:
still an issue as of November 2...

xiulipan, 2018-11-04:
@plbossart 
Waiting for the response about, it seems we did not see similar check from other check tool.

> If Covertiy is some new scan? and how could I re-run to check if this could solve the problem.

tlauda, 2018-11-28:
@plbossart Please verify with #646.

jajanusz, 2018-11-28:
I wonder why tool complained just about 1 line, but it's wrong in 2 places, also here:
https://github.com/thesofproject/sof/blob/4bd543c337615bdd630d0fecef0dced06301a8f9/rimage/pkcs1_5.c#L191

tlauda, 2019-03-25:
@plbossart Could you rescan the project to make sure all the issues are gone?

michalgrodzicki, 2019-03-26:
Task is assigned to @xiulipan, He will do next scan.

xiulipan, 2019-03-28:
@plbossart @tlauda @plbossart 
I will try to run a new round Coverity test on my own machine and to evaluate the possibility to make it into our CI.

jajanusz, 2019-06-06:
Going to be closed with #1524 as initial issue will be fixed.
@xiulipan If you are going to plug Coverity to CI you can create new issue.

closed 2019-06-07 by tlauda

xiulipan, 2019-06-10:
@jajanusz 
The Coverity may not be in CI as it would require third party sever to analysis and take long time for result showing. And the result can not be export for sharing, developer may need access to Coverity to review the test result.
Need to figure out a way to test with Coverity.

PR #1524 (2019-06-07): rimage: fix possible strcpy and sprintf overflows 
It's for preventing possible buffer overflow.

It was partially done in #646 , but missed one occurrence of unsafe strcpy.

Fixes #257