thesofproject/sof #258: rimage: insecure data handling

plbossart, 2018-08-23:
Coverity reports the following issue

```text
46        /* read in sections */
   	3. tainted_data_argument: Calling function fread taints argument section.
 47        count = fread(section, sizeof(Elf32_Shdr), hdr->e_shnum, module->fd);
   	4. Condition count != hdr->e_shnum, taking false branch.
 48        if (count != hdr->e_shnum) {
 49                fprintf(stderr, "error: failed to read %s section header %d\n",
 50                        module->elf_file, -errno);
 51                return -errno;
 52        }
 53
 54        /* read in strings */
   	
CID 313455 (#1 of 2): Untrusted value as argument (TAINTED_SCALAR)
5. tainted_data: Passing tainted variable section[hdr->e_shstrndx].sh_size to a tainted sink.
 55        module->strings = calloc(1, section[hdr->e_shstrndx].sh_size);
```

313455 Untrusted value as argument
The argument could be controlled by an attacker, who could invoke the function with arbitrary values (for example, a very high or negative buffer size).

In elf_read_sections: An unscrutinized value from an untrusted source used as argument to a function (for example, a buffer size) (CWE-20)

tlauda, 2018-11-28:
@plbossart section[hdr->e_shstrndx].sh_size is typedefed as uint32_t, so it cannot be negative nor we shouldn't be bothered with max value. What do you propose to add here? I feel, that checking against negative or zero values is a little bit overkill.

tlauda, 2019-06-06:
@xiulipan Closing as non issue. You can reopen if Coverity will still complain.

closed 2019-06-06 by tlauda

jajanusz, 2019-06-06:
@plbossart @tlauda May be marked as false positive in Coverity. Someone can hack elf section header to have big value 4 fun and it will only result in calloc failing to allocate memory, or crashing rimage. Non-security issue. Doing sanity checks for things like this will only make code less readable.