thesofproject/sof #263: Failed to load sof on UP^2 with GLK-003-drop branch

ZhendanYang, 2018-08-27:
Summary: 
Failed to load sof on UP^2 with GLK-003-drop branch.

Environment:
on up^2 using gcc/xcc
sof: glk-003-drop-stable 3093cf65872eb30c32a621b87be76b39c820e6ce
soft: master 3093cf65872eb30c32a621b87be76b39c820e6ce
sof-dev: 7481f7af7ac711504ae52fa9ae5cb5c3322ea91f
tplg: sof-apl-nocodec.tplg
Log:
[dmesg.log](https://github.com/thesofproject/sof/files/2322309/dmesg.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2322310/rmbox-t.log)

lgirdwood, 2018-08-27:
dmesg shows SOF has loaded ? Playback working too.

keyonjie, 2018-08-27:
dmesg also shows it loading cnl topology?

keyonjie, 2018-08-28:
@ZhendanYang does expanding 256B block number fixing this?

ZhendanYang, 2018-08-28:
@keyonjie 
Expanding 256B block number from 64 to 128 can fix this issue.

keyonjie, 2018-08-28:
@ZhendanYang good to know that, thanks.

keyonjie, 2018-08-28:
@ZhendanYang @keqiaozhang can either of you do a patch to fix it and send PR?

keqiaozhang, 2018-08-28:
@keyonjie 
There's already a patch in sof master, so we need to cherry-pick it to glk-003-drop branch.
Commit ID : thesofproject/sof/commit/fc3ed5eaef4b928cfe08dd6f7f96831638312c72

Jocelyn-Li, 2018-08-28:
@mmaka1 can you please apply the patch commit ID : fc3ed5e to this branch?

wenqingfu, 2018-08-29:
'git branch --contains fc3ed5 ' shows it's been included in glk-003-drop-stable.  

I'm confused.

keqiaozhang, 2018-08-30:
Yes, this commit it included in glk-003-drop-stable before, but after after some mergers, this patch is missed or changed as before.
I used git blame but failed to get the information for this change.
I also tried git bisect, but lead to a wrong commit.
Anywhere, we need to change the HEAP_RT_COUNT256 to 128 for UP^2. or please Xiuli to send another PR to glk-003-drop-stable branch.

mmaka1, 2018-09-01:
glk-003 is for glk fixes only atm

ZhendanYang, 2018-09-10:
This issue still exist on latest code base:
sof: glk-003-drop-stable 13ac4f95bd2cfb3e626d9d9d476e6f0f4289a58c
soft: master d77beaf8bd5260058c00a3fe8f56940eb0962498
kernel: 66dc735d45f64fcfc868c3772bb2c4c52446c000
This issue is a blocker since daily regression test can't be peformed for GLK-003 branch.

mengdonglin, 2018-09-17:
Close this issue because all generic fixes on glk-003 branch have been merged by master.  So testing with master branch will be okay for mature platforms. glk release branch does not focus on compatibility across platforms. 
@ZhendanYang please reopen it if this issue can be reproduced by master branch.

closed 2018-09-17 by mengdonglin