thesofproject/sof #272: Stress test : Keeping press space key when playback causes DSP panic.

ZhendanYang, 2018-08-29:
Summary:
Keep pressing space key when playback cause ipc error, dsp panic, ipc time out.

Repreduce steps:
1. aplay -D hw:0,0 -c 2 -r 48000 -f s16_le music.wav --period-size=192 --buffer-size=384 -vv -i
2. press space key until input/output error
3. dmesg

Environment:
sof-dev : f9c923e5ca9b38df7ddc71f26b55eb2bd1ffed46
sof ： 7346243e596cd91ff4342308e680392600496b3a
soft : 41861aac373eb1f6978bfdb0bbe97cd882b1785a
tplg : 
sof-apl-nocodec.tplg on apl
test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg on byt
test-ssp0-mclk-0-I2S-volume-s16le-s16le-48k-24000k-nocodec.tplg on cnl

When I git revert commit df0d70150ec7c99de128e219e4aef72fc94ec1b9 in sof-master, this issue disappears.

Logs:
on up^2:
[dmesg-apl.log](https://github.com/thesofproject/sof/files/2331316/dmesg-apl.log)
[rmbox-apl.log](https://github.com/thesofproject/sof/files/2331317/rmbox-apl.log)
[rmbox-t-apl.log](https://github.com/thesofproject/sof/files/2331318/rmbox-t-apl.log)

on cnl:
[dmesg-cnl.log](https://github.com/thesofproject/sof/files/2331321/dmesg-cnl.log)
[rmbox-cnl.log](https://github.com/thesofproject/sof/files/2331322/rmbox-cnl.log)
[rmbox-t-cnl.log](https://github.com/thesofproject/sof/files/2331323/rmbox-t-cnl.log)

on byt:
[dmesg-byt.log](https://github.com/thesofproject/sof/files/2331330/dmesg-byt.log)
[rmbox-byt.log](https://github.com/thesofproject/sof/files/2331331/rmbox-byt.log)
[rmbox-t-byt.log](https://github.com/thesofproject/sof/files/2331332/rmbox-t-byt.log)

lgirdwood, 2018-08-29:
It's a (NULL + 0x40) pointer deref at run level 5, probably in the scheduler.

xiulipan, 2018-08-29:
@lgirdwood 
Yes, this panic bugs seems come back.
Since we implement IPC message with scheduler I believe this is related to scheduler.

lgirdwood, 2018-08-29:
@xiulipan I've spotted a possible race we need to resolve in scheduler. We need to add a check to make sure we are not appending a task to the list twice. We should check the schedule list for the task and only re append the task if it's not present. Atm we add the task to the list regardless in _schedule_task().

xiulipan, 2018-08-30:
@lgirdwood 
Yes, this is what @RanderWang and I have found yesterday. But test with the patch seems have no help to this regression.
We also find that `edf_get_next` and reschedule have some problem.

```
static inline struct task *edf_get_next(uint64_t current,
	struct task *ignore)
{
...
int reschedule = 0;
...
	} else {
			/* missed scheduling - will be rescheduled */
			trace_pipe_error("ed!");

			/* have we already tried to rescheule ? */
			if (reschedule++) // What is this check for? If reschedule is 0 we should reschedule?
				edf_reschedule(task, current);
			else {
				/* reschedule failed */
				list_item_del(&task->list);
				task->state = TASK_STATE_CANCEL;
			}
		}
	spin_unlock_irq(&sch->lock, flags);
	return next_task;
}

```

xiulipan, 2018-08-30:
@lgirdwood 

Use XCC we get some panic that seems we have idle for too long time.

```diff
@@ -283,7 +283,15 @@ static void queue_reschedule(struct work_queue *queue)
 static void queue_run(void *data)
 {
        struct work_queue *queue = (struct work_queue *)data;
+       struct list_item *wlist;
        uint32_t flags;
+
+       trace_error(TRACE_CLASS_WAIT ,"qrr");
+       list_for_item(wlist, &queue->work) {
+               trace_error_value((uint32_t)wlist);
+       }
```
> 0xf10 [83.166189]       delta [0.734977]        wait qrr // log in queue_run. IRQ happens.
> 0xf20 [83.166190]       delta [0.000001]        value 0x00000000be046898
> 0xf30 [83.166192]       delta [0.000001]        value 0x00000000be046abc
> 0xf40 [83.908383]       delta [0.742191]        wait qrr
> 0xf50 [83.908384]       delta [0.000001]        value 0x00000000be046898
> 0xf60 [83.908385]       delta [0.000001]        value 0x00000000be046abc
> 0xf70 [83.908390]       delta [0.000005]        sa tim //idle timeout?
> 0xf80 [83.908391]       delta [0.000001]        value 0x0000000000d9701e

I think we may hint this bug because we are not in WFI of the `src/tasks/audio.c` But in `wait_for_completion_timeout(&dd->complete);` in `dai_comp_trigger`.

lgirdwood, 2018-08-30:
@xiulipan scheduler runs at highest IRQ level, so it cant be preempted by dai_trigger(). It can schedule dai_trigger() to be run, but at a lower IRQ level.

Can you try the following.
```diff
diff --git a/src/lib/schedule.c b/src/lib/schedule.c
index b98934d2..8eb84c09 100644
--- a/src/lib/schedule.c
+++ b/src/lib/schedule.c
@@ -197,6 +197,7 @@ static struct task *schedule_edf(void)
                /* yes, run current task */
                task->start = current;
                task->state = TASK_STATE_RUNNING;
+               list_item_del(&task->list);
                arch_run_task(task);
        }
```

can you also change the scheduler error trace to use trace_error() too.

```code
 Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   11.511728] sof-audio sof-audio: EXCCAUSE 0x0000000d EXCVADDR 0x00000040 PS       0x00070025 
```
The NULL pointer deref is always at offset 0x40. What happens if you add some padding uint32_t to the start of the scheduler and workq structures. Does the 0x40 change ?

Finally is the EPC5 program counter value consistent in the same function ?

lgirdwood, 2018-08-30:
@xiulipan agreed, workq may have the error instead or alongside the scheduler.

lgirdwood, 2018-08-30:
@xiulipan please also add locking around this list too 
```code
/* intentionally don't lock list to have task added from schedule irq */
	list_for_item(tlist, &irq_task->list) {
```
This list iterator can be preemptied by the scheduler.

xiulipan, 2018-08-30:
@lgirdwood 
I have tried to lock this and make all thing related atomic, but seems there are several bugs that cause the final result. I have take notes and will try this. Thanks for reminding.

lgirdwood, 2018-08-30:
@xiulipan thanks, btw - please dont wait to send all patches as one PR. i.e. patch for locking irq list is a bug fix so should be sent asap in it's own PR to allow testing.

tlauda, 2018-08-30:
@lgirdwood @xiulipan If we want to lock this list, then you need to rewrite the whole scheduling process. Right now locking the list potentially will block scheduling of the task with the same priority, if the interrupt will be set during task execution.

lgirdwood, 2018-08-30:
@tlauda @xiulipan @RanderWang  I was thinking locking like this

```diff
diff --git a/src/arch/xtensa/task.c b/src/arch/xtensa/task.c
index 98e01eb5..9dbbd8fb 100644
--- a/src/arch/xtensa/task.c
+++ b/src/arch/xtensa/task.c
@@ -71,19 +71,28 @@ static inline uint32_t task_get_irq(struct task *task)
 static inline void task_set_data(struct task *task)
 {
        struct list_item *dst = NULL;
+       struct irq_task *irq_task;
+       uint32_t flags;
+
        switch (task->priority) {
        case TASK_PRI_MED + 1 ... TASK_PRI_LOW:
-               dst = &irq_low_task->list;
+               irq_task = irq_low_task;
+               dst = &irq_task->list;
                break;
        case TASK_PRI_HIGH ... TASK_PRI_MED - 1:
-               dst = &irq_high_task->list;
+               irq_task = irq_high_task;
+               dst = &irq_task->list;
                break;
        case TASK_PRI_MED:
        default:
-               dst = &irq_med_task->list;
+               irq_task = irq_med_task;
+               dst = &irq_task->list;
                break;
        }
+
+       spin_lock_irq(&irq_task->lock, flags);
        list_item_append(&task->irq_list, dst);
+       spin_unlock_irq(&irq_task->lock, flags);
 }
 
 static void _irq_task(void *arg)
@@ -94,18 +103,19 @@ static void _irq_task(void *arg)
        struct task *task;
        uint32_t flags;
 
-       /* intentionally don't lock list to have task added from schedule irq */
-       list_for_item(tlist, &irq_task->list) {
-               task = container_of(tlist, struct task, irq_list);
+       spin_lock_irq(&irq_task->lock, flags);
+       list_for_item_safe(clist, tlist, &irq_task->list) {
+
+               spin_unlock_irq(&irq_task->lock, flags);
+               task = container_of(clist, struct task, irq_list);
 
                if (task->func)
                        task->func(task->data);
 
                schedule_task_complete(task);
+               spin_lock_irq(&irq_task->lock, flags);
        }
 
-       spin_lock_irq(&irq_task->lock, flags);
-
        list_for_item_safe(clist, tlist, &irq_task->list) {
                task = container_of(clist, struct task, irq_list);
                list_item_del(&task->irq_list);
```

This was the list is always locked but the task runs without holding the lock.

xiulipan, 2018-08-31:
@lgirdwood 
I am still testing. I am not sure if this is a fix or not. I am afraid we will involve new regression.
Will send feedback ASAP.

lgirdwood, 2018-08-31:
@xiulipan Can you check task->state prior to running task->func and also set task->state prior to deleteing from list 

```code

if (task->func && task->state == TASK_STATE_RUNNING)
			task->func(task->data);

void schedule_task_complete(struct task *task)
{
	uint32_t flags;

	tracev_pipe("com");

	task->state = TASK_STATE_COMPLETED;
	spin_lock_irq(&sch->lock, flags);
	list_item_del(&task->list);
	spin_unlock_irq(&sch->lock, flags);
}

'''

xiulipan, 2018-08-31:
Here I am not sure if we complete first IPC process. Because the second IPC process comes in from the 
`wait_for_completion_timeout`

```
       case COMP_TRIGGER_PAUSE:
        case COMP_TRIGGER_STOP:
                trace_dai("tsp");
                wait_init(&dd->complete);

                /* wait for DMA to complete */
                dd->complete.timeout = dev->pipeline->ipc_pipe.deadline;
                trace_dai("tsw");
                ret = wait_for_completion_timeout(&dd->complete);
                trace_dai("tsc");
```

From the trace **dai tsw** is called and the paired **dai tsc** is called ever after another IPC with released comes.

0xab960 [34.172782]     delta [0.000794]        ipc IRQ
0xab970 [34.172783]     delta [0.000002]        ipc Nms
0xab980 [34.172787]     delta [0.000003]        pipe sch
0xab990 [34.172792]     delta [0.000006]        pipe pit
0xab9a0 [34.172796]     delta [0.000004]        dai irq
0xab9b0 [34.172800]     delta [0.000004]        pipe sch
0xab9c0 [34.172806]     delta [0.000006]        ipc Cmd
0xab9d0 [34.172809]     delta [0.000003]        ipc tri
0xab9e0 [34.172811]     delta [0.000002]        **ipc trp**
0xab9f0 [34.172812]     delta [0.000001]        pipe cmd
0xaba00 [34.172815]     delta [0.000002]        host trg
0xaba10 [34.172817]     delta [0.000002]        volume trg
0xaba20 [34.172819]     delta [0.000002]        dai trg
0xaba30 [34.172821]     delta [0.000002]        dai tsp
0xaba40 [34.172822]     delta [0.000002]        **dai tsw**
0xaba50 [34.172828]     delta [0.000006]        pipe pit
0xaba60 [34.172829]     delta [0.000002]        ipc Cmd
0xaba70 [34.172832]     delta [0.000003]        ipc tri
0xaba80 [34.172834]     delta [0.000002]        **ipc trp**
0xaba90 [34.172835]     delta [0.000001]        pipe cmd
0xabaa0 [34.172837]     delta [0.000002]        host trg
0xabab0 [34.172839]     delta [0.000002]        comp CEp
0xabac0 [34.172840]     delta [0.000001]        value 0x0000000000000004
0xabad0 [34.172841]     delta [0.000001]        ^[[31mpipe eOp^[[0m
0xabae0 [34.172843]     delta [0.000001]        ipc pc0
0xabaf0 [34.172844]     delta [0.000001]        value 0x0000000000000000
0xabb00 [34.172845]     delta [0.000001]        value 0x0000000000000002
0xabb10 [34.172846]     delta [0.000001]        ^[[31mipc eRc^[[0m
0xabb20 [34.172847]     delta [0.000001]        value 0x0000000000000006
0xabb30 [34.172953]     delta [0.000106]        ipc IRQ
0xabb40 [34.172955]     delta [0.000002]        ipc Nms
0xabb50 [34.172958]     delta [0.000003]        pipe sch
0xabb60 [34.172963]     delta [0.000006]        pipe pit
0xabb70 [34.172965]     delta [0.000002]        ipc Cmd
0xabb80 [34.172967]     delta [0.000002]        ipc tri
0xabb90 [34.172969]     delta [0.000002]        **ipc trr**
0xabba0 [34.172970]     delta [0.000001]        pipe cmd
0xabbb0 [34.172972]     delta [0.000002]        host trg
0xabbc0 [34.172974]     delta [0.000002]        volume trg
0xabbd0 [34.172976]     delta [0.000002]        dai trg
0xabbe0 [34.172978]     delta [0.000002]        ^[[31mdma eS0^[[0m
0xabbf0 [34.172980]     delta [0.000001]        value 0x0000000000000001
0xabc00 [34.172981]     delta [0.000001]        value 0x0000000080000023
0xabc10 [34.172982]     delta [0.000001]        value 0x0000000000000005
0xabc20 [34.172984]     delta [0.000001]        ^[[31mpipe eOp^[[0m
0xabc30 [34.172985]     delta [0.000001]        ipc pc0
0xabc40 [34.172986]     delta [0.000001]        value 0x0000000000000000
0xabc50 [34.172987]     delta [0.000001]        value 0x0000000000000003
0xabc60 [34.172988]     delta [0.000001]        ^[[31mipc eRc^[[0m
0xabc70 [34.172989]     delta [0.000001]        value 0x0000000000000007
0xabc80 [34.173796]     delta [0.000807]        dai irq
0xabc90 [34.173799]     delta [0.000003]        ^[[31mdai xru^[[0m
0xabca0 [34.173800]     delta [0.000001]        ^[[31mcomp Xun^[[0m
0xabcb0 [34.173802]     delta [0.000001]        value 0x0000000000040000
0xabcc0 [34.173803]     delta [0.000002]        value 0x00000000000000c0
0xabcd0 [34.173810]     delta [0.000007]        pipe sch
0xabce0 [34.173817]     delta [0.000007]        pipe pit
0xabcf0 [34.173819]     delta [0.000002]        pipe pxr
0xabd00 [34.173821]     delta [0.000001]        pipe cmd
0xabd10 [34.173822]     delta [0.000002]        host trg
0xabd20 [34.173825]     delta [0.000002]        host DDi
0xabd30 [34.173827]     delta [0.000003]        volume trg
0xabd40 [34.173829]     delta [0.000002]        dai trg
0xabd50 [34.173831]     delta [0.000002]        dai txr
0xabd60 [34.173832]     delta [0.000001]        ssp tri
0xabd70 [34.173835]     delta [0.000003]        ssp Ss1
0xabd80 [34.173837]     delta [0.000002]        ssp Ss2
0xabd90 [34.173839]     delta [0.000002]        dma DDi
0xabda0 [34.173842]     delta [0.000003]        pipe pre
0xabdb0 [34.173844]     delta [0.000002]        host pre
0xabdc0 [34.173846]     delta [0.000002]        volume pre
0xabdd0 [34.173850]     delta [0.000003]        dai pre
0xabde0 [34.173855]     delta [0.000005]        pipe cmd
0xabdf0 [34.173857]     delta [0.000002]        host trg
0xabe00 [34.173859]     delta [0.000002]        host DEn
0xabe10 [34.173895]     delta [0.000036]        volume trg
0xabe20 [34.173897]     delta [0.000002]        dai trg
0xabe30 [34.173898]     delta [0.000002]        dai tsa
0xabe40 [34.173902]     delta [0.000004]        ssp tri
0xabe50 [34.173905]     delta [0.000002]        ssp sta
0xabe60 [34.173908]     delta [0.000004]        pipe tsk
0xabe70 [34.173966]     delta [0.000058]        value 0x0000000000000001
0xabe80 [34.173967]     delta [0.000001]        value 0x0000000000000000
0xabe90 [34.173969]     delta [0.000001]        **dai tsc**

xiulipan, 2018-08-31:
@lgirdwood 
I think we need some check to make sure above situation will not happen.

lgirdwood, 2018-08-31:
@xiulipan please do 

```code
static void _irq_task(void *arg)
{
	struct irq_task *irq_task = *(struct irq_task **)arg;
	struct list_item *tlist;
	struct list_item *clist;
	struct task *task;
	uint32_t flags;

	spin_lock_irq(&irq_task->lock, flags);
	list_for_item_safe(clist, tlist, &irq_task->list) {

		spin_unlock_irq(&irq_task->lock, flags);
		task = container_of(clist, struct task, irq_list);

		if (task->func && task->state == TASK_STATE_RUNNING)
			task->func(task->data);

		schedule_task_complete(task);
		spin_lock_irq(&irq_task->lock, flags);
		list_item_del(&task->irq_list);
	}

	interrupt_clear(irq_task->irq);

	spin_unlock_irq(&irq_task->lock, flags);
}
```

This immediatley removes the task from the list after the func() now.

lgirdwood, 2018-08-31:
@xiulipan can you modify above to remove task from list before we call task->func()

lgirdwood, 2018-08-31:
@xiulipan  i.e.
```code
static void _irq_task(void *arg)
{
	struct irq_task *irq_task = *(struct irq_task **)arg;
	struct list_item *tlist;
	struct list_item *clist;
	struct task *task;
	uint32_t flags;

	spin_lock_irq(&irq_task->lock, flags);
	list_for_item_safe(clist, tlist, &irq_task->list) {

		task = container_of(clist, struct task, irq_list);
		list_item_del(&task->irq_list);

		spin_unlock_irq(&irq_task->lock, flags);

		if (task->func && task->state == TASK_STATE_RUNNING)
			task->func(task->data);

		schedule_task_complete(task);
		spin_lock_irq(&irq_task->lock, flags);
	}

	interrupt_clear(irq_task->irq);

	spin_unlock_irq(&irq_task->lock, flags);
}
```

xiulipan, 2018-08-31:
@lgirdwood 
I will try to fix this bug, since I am now root caused the risk situation.
I thing remove it first may have some help. I will have some test and try to send the PR.

xiulipan, 2018-08-31:
@lgirdwood 
Test with the patches above.
The first IPC send to the DSP is timeout.

> [    3.712396] sof-audio sof-audio: error: ipc timed out for 0x90010000 size 0x1c
> [    3.712455] sof-audio sof-audio: status: fw entered - code 00000005
> [    3.712507] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
> [    3.712552] sof-audio sof-audio: error: can't set params for DMA for trace -110

xiulipan, 2018-09-03:
#292 will fix unhanded IPC issues part that caused by the list chaos.
Also need the arch task refine to avoid run same IPC process twice.
```c
if (task->func && task->state == TASK_STATE_RUNNING)
			task->func(task->data);
```

xiulipan, 2018-09-03:
> 0x0 [41.345327] delta [41.345327]       dai xru // happen at pause last one dai cb
> 0x10 [41.363386]        delta [0.018059]        dai xru // happen at release dai cb
> 0x20 [41.363402]        delta [0.000017]        pipe pxr // cause noise
> 0x30 [305.105971]       delta [263.742569]      dai xru
> 0x40 [305.120735]       delta [0.014764]        dai xru
> 0x50 [305.120751]       delta [0.000016]        pipe pxr
> 0x60 [308.867960]       delta [3.747209]        dai xru
> 0x70 [308.882283]       delta [0.014323]        dai xru
> 0x80 [308.882299]       delta [0.000016]        pipe pxr

The noise will recover at another **dai xru**

xiulipan, 2018-09-03:
Pretty sure is caused by xrun recovery pointer chaos.

xiulipan, 2018-09-03:
part patch about arch task #298

RanderWang, 2018-09-05:
@lgirdwood  @xiulipan 
xiuli has sent a report to describe the strange bug, and I update my finding for I suddently got a clear picture on my way home.  

The kernel panic is caused by IPC schedule change. It is a exception in FW at
ipc_cmd
     dai_comp_trigger
     {
        case PAUSE: 
         inline wait_for_complete()
        {
                 ....
                register a4(store the correct address of dev_comp)
                arch_wait_for_interrupt(0)
                a4 = 8 // after wait
                l32i a6, a4, 56 // exception here when accessing the content of dev_comp
         }
   }

tlauda, 2018-09-05:
@RanderWang @xiulipan @lgirdwood 
IPC's are no longer executed on passive level, so arch_wait_for_interrupt(0) lowers IPC task level to 0 and other tasks are triggering task execution irq level and causing stack overflow. PR #284 should fix this issue, but I think it indicates that the scheduler has still problem with locking task list.

RanderWang, 2018-09-05:
It is a strange behavior!  I dumped all the a0 - a15 before and after wait_interrupt() and have interesting finding. many registers are changed, and two special register are useful,  one is return address register a0, another is a data register a7.

           a0 = correct return address of dai_comp_trigger

           a7 = correct data for this function

            wait_for_interrupt(0)

            a0 = Panic(5) at the end of main, so it should return to main !!!

             a7 = data used in main()

It is a great jump!  But my idea it is that the content of these registers are for task_run() for its return address is panic(5). And there is a wait_for_interrupt(0) in task_run() also. what would happen if two wait_for_interrupt are waiting for interrupt simultaneously in different interrupt level, then a dma interrupt comes ? this maybe make a chaos like this.

RanderWang, 2018-09-05:
@tlauda  @lgirdwood @xiulipan 

hi Tomasz,  xiuli has tried your patch, still kernel panic

lgirdwood, 2018-09-05:
@tlauda sounds like we need a check in WFI that returns an error if run level is not 0
@ZhendanYang can you please check PR #284, we need this checked before merging.

lgirdwood, 2018-09-05:
@RanderWang @xiulipan  Please try the following patch and fix any subsequent panics.
```diff
diff --git a/src/arch/xtensa/include/arch/wait.h b/src/arch/xtensa/include/arch/wait.h
index e1cca01c..8f443e5c 100644
--- a/src/arch/xtensa/include/arch/wait.h
+++ b/src/arch/xtensa/include/arch/wait.h
@@ -29,6 +29,7 @@
  */
 
 #include <xtensa/xtruntime.h>
+#include <arch/interrupt.h>
 
 #if defined(PLATFORM_WAITI_DELAY)
 
@@ -36,6 +37,10 @@ static inline void arch_wait_for_interrupt(int level)
 {
        int i;
 
+       /* can only eneter WFI when at runlevel 0 i.e. not IRQ level */
+       if (arch_interrupt_get_level() > 0)
+               panic(SOF_IPC_PANIC_WFI);
+
        /* this sequnce must be atomic on LX6 */
        XTOS_SET_INTLEVEL(5);
 
@@ -55,7 +60,11 @@ static inline void arch_wait_for_interrupt(int level)
 
 static inline void arch_wait_for_interrupt(int level)
 {
-        asm volatile("waiti 0");
+       /* can only eneter WFI when at runlevel 0 i.e. not IRQ level */
+       if (arch_interrupt_get_level() > 0)
+               panic(SOF_IPC_PANIC_WFI);
+
+       asm volatile("waiti 0");
 }
 
 #endif
diff --git a/src/include/uapi/ipc.h b/src/include/uapi/ipc.h
index 68f638e4..1fae8096 100644
--- a/src/include/uapi/ipc.h
+++ b/src/include/uapi/ipc.h
@@ -137,16 +137,17 @@
 #define SOF_IPC_PANIC_MAGIC                    0x0dead000
 #define SOF_IPC_PANIC_MAGIC_MASK               0x0ffff000
 #define SOF_IPC_PANIC_CODE_MASK                        0x00000fff
-#define SOF_IPC_PANIC_MEM                      (SOF_IPC_PANIC_MAGIC | 0)
-#define SOF_IPC_PANIC_WORK                     (SOF_IPC_PANIC_MAGIC | 1)
-#define SOF_IPC_PANIC_IPC                      (SOF_IPC_PANIC_MAGIC | 2)
-#define SOF_IPC_PANIC_ARCH                     (SOF_IPC_PANIC_MAGIC | 3)
-#define SOF_IPC_PANIC_PLATFORM                 (SOF_IPC_PANIC_MAGIC | 4)
-#define SOF_IPC_PANIC_TASK                     (SOF_IPC_PANIC_MAGIC | 5)
-#define SOF_IPC_PANIC_EXCEPTION                        (SOF_IPC_PANIC_MAGIC | 6)
-#define SOF_IPC_PANIC_DEADLOCK                 (SOF_IPC_PANIC_MAGIC | 7)
-#define SOF_IPC_PANIC_STACK                    (SOF_IPC_PANIC_MAGIC | 8)
-#define SOF_IPC_PANIC_IDLE                     (SOF_IPC_PANIC_MAGIC | 9)
+#define SOF_IPC_PANIC_MEM                      (SOF_IPC_PANIC_MAGIC | 0x0)
+#define SOF_IPC_PANIC_WORK                     (SOF_IPC_PANIC_MAGIC | 0x1)
+#define SOF_IPC_PANIC_IPC                      (SOF_IPC_PANIC_MAGIC | 0x2)
+#define SOF_IPC_PANIC_ARCH                     (SOF_IPC_PANIC_MAGIC | 0x3)
+#define SOF_IPC_PANIC_PLATFORM                 (SOF_IPC_PANIC_MAGIC | 0x4)
+#define SOF_IPC_PANIC_TASK                     (SOF_IPC_PANIC_MAGIC | 0x5)
+#define SOF_IPC_PANIC_EXCEPTION                        (SOF_IPC_PANIC_MAGIC | 0x6)
+#define SOF_IPC_PANIC_DEADLOCK                 (SOF_IPC_PANIC_MAGIC | 0x7)
+#define SOF_IPC_PANIC_STACK                    (SOF_IPC_PANIC_MAGIC | 0x8)
+#define SOF_IPC_PANIC_IDLE                     (SOF_IPC_PANIC_MAGIC | 0x9)
+#define SOF_IPC_PANIC_WFI                      (SOF_IPC_PANIC_MAGIC | 0xa)
 
 /*
  * SOF memory capabilities, add new ones at the end
```

lgirdwood, 2018-09-05:
@xiulipan @RanderWang @tlauda  here is the user list for WFI 
```code
src/audio/dai.c:609:		ret = wait_for_completion_timeout(&dd->complete);
src/tasks/audio.c:86:		wait_for_interrupt(0);
src/ipc/hsw-ipc.c:160:			wait_for_interrupt(0);
src/ipc/pmc-ipc.c:149:	wait_for_interrupt(0);
src/ipc/dma-copy.c:142:		err = wait_for_completion_timeout(&dc->complete);
src/ipc/dma-copy.c:296:		err = wait_for_completion_timeout(&dc->complete);
src/ipc/byt-ipc.c:164:			wait_for_interrupt(0);
src/ipc/ipc.c:505:	ret = wait_for_completion_timeout(&complete);
src/ipc/cnl-ipc.c:144:			wait_for_interrupt(0);
src/ipc/apl-ipc.c:142:			wait_for_interrupt(0);
```

xiulipan, 2018-09-06:
@lgirdwood 

- To fix waiti bugs. we need 

#284 #314 

- For list item,
 1. we may need some unification about how to refine the list_item_del. We may need to init every item at each del. #303 or #300
2.  #313 for some potential bugs. 

- WFI check patch set
#315 https://github.com/thesofproject/linux/pull/108

With all these patches now the #272 and #210 should all be fixed.
But now have some xrun inside firmware.
And pipeline xrun recovery will have some noise.
@ZhendanYang @keqiaozhang 
Please report new bugs about noise after all patches are merged and test.

> 0x1de0 [519.288077]     delta [7.678624]        dai xro
> 0x1df0 [519.289141]     delta [0.001064]        dai xro
> 0x1e00 [519.289158]     delta [0.000017]        pipe pxr
> 0x1e10 [526.598085]     delta [7.308927]        dai xro
> 0x1e20 [526.599149]     delta [0.001064]        dai xro
> 0x1e30 [526.599166]     delta [0.000017]        pipe pxr
> 0x1e40 [550.429521]     delta [23.830355]       dai xro
> 0x1e50 [550.430585]     delta [0.001064]        dai xro
> 0x1e60 [550.430601]     delta [0.000017]        pipe pxr
> 0x1e70 [551.177593]     delta [0.746991]        dai xro
> 0x1e80 [551.178657]     delta [0.001064]        dai xro
> 0x1e90 [551.178674]     delta [0.000017]        pipe pxr
>

lgirdwood, 2018-09-06:
@cujomalainey patches above now being merged, should be fully validated next week.

ZhendanYang, 2018-09-07:
This issue can't be reproduced on APL and CNL with stable-1.2 branch(commit 251e68f291853e3c22ce617f23f94c01231268db), need to test on BYT before close this issue.

ZhendanYang, 2018-09-10:
This issue can't be reproduced on APL and CNL with stable-1.2 branch(commit 251e68f291853e3c22ce617f23f94c01231268db)

on APL and CNL with master branch(commit: 7abb55fe722c9a180aa36ad8ed9c8ca906fcc829): no more DSP panic, instead of ipc timeout:

[ 1895.936287] sof-audio sof-audio: error: ipc timed out for 0x60040000 size 0xc
[ 1895.936312] sof-audio sof-audio: status: fw entered - code 00000005
[ 1895.936430] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[ 1895.936436] sof-audio sof-audio: error: waking up any trace sleepers
[ 1895.936460]  Passthrough: ASoC: trigger FE failed -110
[ 1895.936472] sof-audio sof-audio: pcm: trigger stream 0 dir 0 cmd 0
[ 1896.256303] sof-audio sof-audio: error: ipc timed out for 0x60050000 size 0xc
[ 1896.256321] sof-audio sof-audio: status: fw entered - code 00000005
[ 1896.256463] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[ 1896.256468] sof-audio sof-audio: error: waking up any trace sleepers
[ 1896.256487]  Passthrough: ASoC: trigger FE failed -110
[ 1896.604457] sof-audio sof-audio: ipc rx: 0x90020000
[ 1896.604551] sof-audio sof-audio: ipc rx done: 0x90020000
[ 1897.604542] sof-audio sof-audio: ipc rx: 0x90020000
[ 1897.604638] sof-audio sof-audio: ipc rx done: 0x90020000
[ 1898.272253] sof-audio sof-audio: pcm: free stream 0 dir 0
[ 1898.592284] sof-audio sof-audio: error: ipc timed out for 0x60030000 size 0xc
[ 1898.592307] sof-audio sof-audio: status: fw entered - code 00000005
[ 1898.592424] sof-audio sof-audio: error: unexpected fault 0x00000000 trace 0x00004000
[ 1898.592429] sof-audio sof-audio: error: waking up any trace sleepers

Test on BYT is blocked by #274 .
Test on glk-003 branch is blocked by #263 .

mengdonglin, 2018-09-10:
CNL has no DSP panic now. Will create new bug track IPC timeout.

lgirdwood, 2018-09-10:
CNL has different IPC IP compared to APL so could also be driver bug.

mengdonglin, 2018-09-17:
Close this issue because all generic fixes on glk-003 branch have been merged by master.  So testing with master branch will be okay for mature platforms. glk release branch does not focus on compatibility across platforms. 

@ZhendanYang Please re-open this issue if it can be reproduce by master branch and use br-master to track it. Thanks!

PR #298 (2018-09-03): task: refine task to avoid double task function run
Add spinlock to the arch task irq list to avoid double function run.

Have effect to fix #272
closed 2018-09-17 by mengdonglin