thesofproject/sof #274: DSP panic after booting up on BYT.

ZhendanYang, 2018-08-29:
Summary:
DSP panic when baytrail boot up, but this doesn't happen every time. When I disable pulseaudio, this issue disappears. I git revert commit df0d70150ec7c99de128e219e4aef72fc94ec1b9 in sof-master, after that, either I enable or disable pulseaudio, this issue can't be repreduced.

Reproduce steps:
1. boot
2. dmesg

Environment:
on byt
sof-dev : f9c923e5ca9b38df7ddc71f26b55eb2bd1ffed46
sof ： 7346243e596cd91ff4342308e680392600496b3a
soft : 41861aac373eb1f6978bfdb0bbe97cd882b1785a
tplg: test-ssp2-mclk-0-I2S-volume-s16le-s16le-48k-19200k-codec.tplg

Log:
[dmesg-panic.log](https://github.com/thesofproject/sof/files/2331429/dmesg-panic.log)
[rmbox-panic.log](https://github.com/thesofproject/sof/files/2331430/rmbox-panic.log)
[rmbox-t-panic.log](https://github.com/thesofproject/sof/files/2331431/rmbox-t-panic.log)

lgirdwood, 2018-08-29:
duplicate of #272

lgirdwood, 2018-08-30:
duplicate of #210

ZhendanYang, 2018-09-03:
DSP panic can also happen after booting on APL using glk-003 branch sof, either I enable or disable pulseaudio. But kernel panic can happen only after booting when pulseaudio is enabled.
sof : glk-003-drop-stable 0ead9e762b36b87cd59afe44e90f912978a79fad
kernel : bafb59032343459cc4019cb65e515595451357e7
soft : d40087fdd4ecaf0329059ef63527e3d4664b2d59
log:
[dmesg-bootpanic.log](https://github.com/thesofproject/sof/files/2344027/dmesg-bootpanic.log)
[rmbox-bootpanic.log](https://github.com/thesofproject/sof/files/2344028/rmbox-bootpanic.log)
[rmbox-t-bootpanic.log](https://github.com/thesofproject/sof/files/2344029/rmbox-t-bootpanic.log)
[dmesg-kernelpanic.log](https://github.com/thesofproject/sof/files/2344030/dmesg-kernelpanic.log)
[rmbox-kernelpanic.log](https://github.com/thesofproject/sof/files/2344031/rmbox-kernelpanic.log)
[rmbox-t-kernelpanic.log](https://github.com/thesofproject/sof/files/2344032/rmbox-t-kernelpanic.log)

ZhendanYang, 2018-09-07:
DSP panic will occur after boot on BYT either enable or disable pulseaudio using stable 1.2 branch.
This can be a blocker since I can't re-test issue #180 , #210 , #272 .

lgirdwood, 2018-09-07:
@ZhendanYang this is the wait state error code.  @xiulipan are we missing any patches ?

xiulipan, 2018-09-07:
@lgirdwood 
Did not test with BYT. Need to test with it. I think we may miss some WFI for the BYT.

RanderWang, 2018-09-07:
@xiulipan  @lgirdwood  yes, this case is different with APL. The EPCx is different with APL.

RanderWang, 2018-09-07:
@xiulipan @lgirdwood  no such issue on bdw

lgirdwood, 2018-09-07:
@xiulipan ipc-pmc.c on BYT

xiulipan, 2018-09-07:
@lgirdwood @RanderWang 
We need a clean up for all platforms and have a test for this.
Add SOF_IPC_PANIC_WFI is a very good idea to help debug this kind of bug.

ZhendanYang, 2018-09-10:
Can't reproduce on APL and CNL with stable-1.2 branch firmware either pulseaudio is enabled or disabled. 
But can reproduce on BYT with stable-1.2 branch firmware either pulseaudio is enabled or disabled. 
This is a blocker for validating other issues.

xiulipan, 2018-09-10:
@ZhendanYang 
@RanderWang have PRs for BYT #333

ZhendanYang, 2018-09-14:
Latest status:
Test with stable1.2 branch(f7ea2e61606f78a047a7353cb8042242c93448dc), enable and disable pulseaudio both will cause DSP panic, but seems the reproducibity is much less than before.
Test with master branch(f17abe644c3b583213914b3d98e27f55786ef8c7), DSP panic occurs every time whether enable pulseaudio or not.

lgirdwood, 2018-09-14:
@ZhendanYang is the reason still [ 8.249425] sof-audio sof-audio: error: unknown reason 0dead00a ??

This is the wait checker we have to detect incorrect wait transition states

ZhendanYang, 2018-09-17:
@lgirdwood 
No, the reason changed.
With stable-1.2 branch, when pulseaudio enabled :
[    8.214583] sof-audio sof-audio: error : DSP panic!
[    8.214594] sof-audio sof-audio: panic: dsp_oops_offset 1329152 offset 1329152
[    8.214624] sof-audio sof-audio: error: invalid wait state
[    8.214630] sof-audio sof-audio: error: trace point 90010000
[    8.214636] sof-audio sof-audio: error: DSP Firmware Oops
[    8.214641] sof-audio sof-audio: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[    8.214649] sof-audio sof-audio: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00060125 SAR     0x00000011
[    8.214657] sof-audio sof-audio: EPC1     0xff2c29f9 EPC2     0xff2c0f1b EPC3     0xff2c0f1b EPC4    0xff2c0f1b
[    8.214665] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    8.214672] sof-audio sof-audio: EPS2     0x00060520 EPS3     0x00060520 EPS4     0x00060520 EPS5    0x00000000
[    8.214680] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x0007ef48 INTERRU 0x000020a2
[    8.214687] sof-audio sof-audio: stack dump from 0xff327d30
[dmesg-1.2-enable.log](https://github.com/thesofproject/sof/files/2387245/dmesg-1.2-enable.log)

WIth master branch, when pulseaudio disabled :
[    7.898558] sof-audio sof-audio: error : DSP panic!
[    7.898596] sof-audio sof-audio: error: out of memory
[    7.898603] sof-audio sof-audio: error: trace point 00004170
[    7.898608] sof-audio sof-audio: error: DSP Firmware Oops
[    7.898614] sof-audio sof-audio: error: Exception Cause: IllegalInstructionCause, Illegal instruction
[    7.898622] sof-audio sof-audio: EXCCAUSE 0x00000000 EXCVADDR 0x00000000 PS       0x00060525 SAR     0x00000011
[    7.898630] sof-audio sof-audio: EPC1     0xff2ce8ed EPC2     0x00000000 EPC3     0x00000000 EPC4    0xff2cce8f
[    7.898637] sof-audio sof-audio: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[    7.898645] sof-audio sof-audio: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00060720 EPS5    0x00000000
[    7.898653] sof-audio sof-audio: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00008b48 INTERRU 0x000000a2
[    7.898659] sof-audio sof-audio: stack dump from 0xff327f20
[dmesg-disable-master.log](https://github.com/thesofproject/sof/files/2387243/dmesg-disable-master.log)

xiulipan, 2018-09-17:
@ZhendanYang 

> [ 8.214624] sof-audio sof-audio: error: invalid wait state

is same as 

> [ 8.249425] sof-audio sof-audio: error: unknown reason 0dead00a

@lgirdwood 
We still missing one PR https://github.com/thesofproject/sof/pull/335
to remove all waiti on BYT

And we need to enable another DMA stop just like #342 

With updated #335 and #342, I think BYT could be good.

xiulipan, 2018-09-18:
@ZhendanYang 
#335 and #342 is merged on stable-1.2
Please verify.

I will send merge PR from stabe-1.2 to master.

ZhendanYang, 2018-09-18:
Verified. This issue can't reproduce on stable-1.2, but still exists on master.

lgirdwood, 2018-09-20:
@xiulipan the master branch out of memory may need heap size adjustment.. Btw it would be nice if panic() also included line number and filename in the dump it gives to the kernel.

xiulipan, 2018-09-21:
@lgirdwood 
OK, I create a issues https://github.com/thesofproject/sof/issues/403 to track this

xiulipan, 2018-09-26:
@lgirdwood 
Now panic info patch set is test with BYT and APL, here is the result on BYT

[    5.685890] sof-audio sof-audio: error : DSP panic!
[    5.686049] sof-audio sof-audio: error: out of memory
[    5.686132] sof-audio sof-audio: error: trace point 00004170
[    5.686220] sof-audio sof-audio: error: panic happen at alloc.c:131

```c
128         /* always succeeds or panics */
129         if (alignment + bytes > cpu_heap->info.free) {
130                 trace_mem_error("eM1");
131                 panic(SOF_IPC_PANIC_MEM);
132         }
```

Will send the PR first and start debut this issue.

PR #424 (2018-09-26): byt: enlarge system heap size
With some modification in memory alloction, system heap used is enlarged
from 0x16fc to 0x2fcc.

Modify the system heap size to 0xe000 to align with other platforms.

close #274
commit a0b169d5 (2018-09-26): byt: enlarge system heap size

With some modification in memory alloction, system heap used is enlarged
from 0x16fc to 0x2fcc.

Modify the system heap size to 0xe000 to align with other platforms.

close #274
closed 2018-09-26 by lgirdwood