thesofproject/sof #276: Failed to load nocodec tplg on UP^2 with master branch

keqiaozhang, 2018-08-30:
Failed to loadsof-apl-nocodec.tplg  on UP^2.  but pcm512 codec tplg can be loaded and audio works well.
Trace log：
0xe30 [0.069816]        delta [0.000028]        dai new
0xe40 [0.069824]        delta [0.000008]        mem eMF
0xe50 [0.069827]        delta [0.000002]        comp eCN
0xe60 [0.069828]        delta [0.000001]        ipc eCn
0xe70 [0.069829]        delta [0.000001]        ipc cn1

Test config: sof-dev + sof-master + soft-master(both gcc and xcc)
sof-dev: bafb590
sof-master: 9944905
soft-master: 41861aa
tplg name: sof-apl.nocodec.tplg

[dmesg.log](https://github.com/thesofproject/sof/files/2334862/dmesg.log)
[rmbox-t.log](https://github.com/thesofproject/sof/files/2334863/rmbox-t.log)

lgirdwood, 2018-08-30:
@ZhendanYang master bugs are P3 atm as it's in development. stable-1.2 bugs have priority.

ZhendanYang, 2018-09-10:
This issue can't be reproduced so close it.

closed 2018-09-10 by ZhendanYang